Xilinx/Vitis-AI #180: Vitis AI Compiler for Caffe model error #3

JerrySciangula, 2020-10-14:
Hi everyone,

I'm a student at Scuola Superiore Sant'Anna in Pisa.

I'm trying to accelerate a network model on a ZCU102 ultrascale+.
I can successfully quantize the model using vai_q_caffe and the tool correctly generates the 4 quantize_results files.

When I try to compile the model using vai_c_caffe, it outputs:

**[VAI_C][Warning] Only 'channel' axis supported on DPU for Concat, current layer is [Concat1].
[VAI_C][Fatal] Check failed for condition [slice_point.size() == current->out_value_num()] in [/home/xbuild/conda-bld/dnnc_1592904456005/work/dnnc_impl/parser/caffe/caffeparser.cc:1325] :**

Could anyone explain me the error and how to fix it, if it is possible?

Mookel, 2020-10-15:
Hi, @JerrySciangula 

Could you share your deploy.prototxt, thus we can analysis your issue further?
Thanks very much.

JerrySciangula, 2020-10-15:
Hi, @Mookel

Thank you for your reply! 

The deploy.prototxt file can be reached at this link: https://github.com/lgsvl/apollo-5.0/blob/simulator/modules/perception/production/data/perception/camera/models/lane_detector/darkSCNN/deploy.prototxt

JerrySciangula, 2020-10-19:
@Mookel, I'm reposting the prototxt also here: 

`name: "darknet-16c-16x-3d multitask"

# SCNN part: kernel size 5, only Up-Down direction
###################### LANE #######################3

layer {
  name: "data"
  type: "ImageData"
  top: "data"
  top: "label"

  transform_param {
    mean_value: 95
    mean_value: 99
    mean_value: 96
    mirror: false
  }

  image_data_param {
    root_folder: "./sample_image/reshaped_t/"
    source: "./sample_image/calibration_t.txt"
    batch_size: 1
    shuffle: true
    new_height: 480
    new_width: 640
    #################################
    scale: 0.00392157
    #################################
  }
  include {
    phase: TRAIN
  }
}

layer {
  name: "conv1"
  type: "Convolution"
  bottom: "data"
  top: "conv1"
  param {
    lr_mult: 1
    decay_mult: 1
  }
  convolution_param {
    num_output: 16
    bias_term: false
    pad: 1
    kernel_size: 3
    dilation: 1
    stride: 1
    weight_filler {
      type: "gaussian"
      std: 0.01
    }
  }
}
layer {
  name: "conv1_bn"
  type: "BatchNorm"
  bottom: "conv1"
  top: "conv1"
  param {
    lr_mult: 1
    decay_mult: 1
  }
  param {
    lr_mult: 1
    decay_mult: 1
[... 6461 lines cut ...]
  bottom: "fc1"
  top: "fc_out"
  # learning rate and decay multipliers for the weights
  param {
    lr_mult: 1
    decay_mult: 1
  }
  # learning rate and decay multipliers for the biases
  param {
   lr_mult: 1
    decay_mult: 0
  }
  inner_product_param {
    num_output: 2
    weight_filler {
      type: "gaussian"
      std: 0.01
    }
    bias_filler {
      type: "constant"
      value: 0
    }
  }
}
`

Mookel, 2020-10-20:
Hi, @JerrySciangula 

I checked the code and found a bug for the compiler: our compiler requires that the the indices must be equal to the top blobs, while the page from Caffe indicates that the number of indices of the Silce must be equal to the number of top blobs minus one(https://caffe.berkeleyvision.org/tutorial/layers/slice.html). Thanks you very much and we will fixed it later.

Another fact I would like to point out is that: even this bug is fixed, you have to implement the Slice layer by yourself using C++ in the deployment code, the reason is that the Slice layer can not be supported by our DPU and the compiler will partiton this model into different pieces and insert necessary input layer for them. So before the bug is fixed and relying on the compiler to partition the models, you can have a try and do it manully.

Hope it helps and thanks very much.

JerrySciangula, 2020-10-20:
Hi @Mookel, 

thank you very much for your reply! I think the issue can be closed.

Have a nice day.

closed 2020-10-20 by JerrySciangula