Xilinx/Vitis-AI #181: General question about DPU

kevinkit, 2020-10-15:
When Looking at the DPU Ip Core i noticed that the DPU has one AXI Slave Input and three outputs, where one of these outputs is titled "DPU0_M_AXI_INSTR" - I suppose, that INSTR means INSTRUCTIONS - however, if the DPU is thought as Co-Prozessor and the Instructions are coming from the APU - why isn't there a Slave axi for Instructions, and why has the DPU a Master instruction axi ? Or is the MASTER INSTR output something different?

Thank you for your help

foolmarks, 2020-10-15:
The DPU reads its instructions from the compiled model file (.elf for Zynq, .model for Alveo) which is in memory.It also reads and writes data to external memory.Sent from my Samsung Galaxy smartphone.

kevinkit, 2020-10-15:
Thank you - so what is the INSTR Axi port for?

JasonXijieJia, 2022-02-21:
Hi @kevinkit 
  The instructions are stored in DDR statically by the runtime. Then the master AXI INSTR port fetches the instructions from ddr in dma mode. 
  Thus the cpu could be released for other threads.

closed 2022-03-03 by qianglin-xlnx