Xilinx/Vitis-AI #182: Vitis AI Quantizer for Caffe model error

JerrySciangula, 2020-10-15:
Hi everyone,

I'm a student at Scuola Superiore Sant'Anna in Pisa.

I'm trying to accelerate a network model on a ZCU102 ultrascale+. Here the prototxt of the model: https://github.com/ApolloAuto/apollo/blob/master/modules/perception/production/data/perception/camera/models/traffic_light_recognition/vertical/deploy.prototxt. 

I have just the deploy.prototxt file and not the train.prototxt so I substituted the Input layer with an ImageData layer, in order to execute the calibration of the network.

When I try to quantize the model using vai_q_caffe, it outputs:

**convert_proto.cpp:2024] Check failed: sc_param.blobs_size() == 2 (1 vs. 2)  Wrong ScaleLayer blob size of layer: conv1_bn_scale**

Could anyone explain me the error and how to fix it, if it is possible?

kevinkit, 2020-10-21:
The batchnorm layer will get "integrated" into the convolutional layer, I am not quite sure about your architecture but maybe there is some kind of split or a layer between the conv and the batch norm layer that hinders this quantization step

JerrySciangula, 2020-10-21:
This is the very first part of the prototxt file:

layer{
  name: "input"
  type: "Input"
  top: "data_org"
  input_param{
    shape {
      dim:1
      dim:96
      dim:32
      dim:3
    }
  }
}

layer {
    type: "Permute"
    name: "permute"
    bottom: "data_org"
    top: "data"
    permute_param{
        order: 0
        order: 3
        order: 1
        order: 2
    }
}

layer{
    name: "conv1"
    type: "Convolution"
    bottom: "data"
    top: "conv1"
    param{
        lr_mult: 1.000000
        decay_mult: 1.000000
    }
    param {
        lr_mult: 2.000000
        decay_mult: 0.000000
    }
    convolution_param {
      num_output: 32
      kernel_size: 3
      pad: 1
      stride: 1
      dilation: 1
      weight_filler {
        type: "msra"
      }
      bias_filler {
        type: "constant"
        value: 0.000000
      }
    }
}
layer{
  type: "BatchNorm"
  name: "conv1_bn"
  bottom: "conv1"
  top: "conv1"
  batch_norm_param{
    use_global_stats: true
  }
}
layer {
  type: "Scale"
  name: "conv1_bn_scale"
  bottom: "conv1"
  top: "conv1"
  scale_param {
    axis: 1
    num_axes: 1
    bias_term: false
  }
}
layer{
  type: "ReLU"
  name: "conv1_relu"
  bottom: "conv1"
  top: "conv1"
}
layer {
  name: "pool1"
  type: "Pooling"
  bottom: "conv1"
  top: "pool1"
  pooling_param {
    pool: MAX
    kernel_w: 3
    kernel_h: 3
    stride_w: 2
    stride_h: 2
    pad_w: 1
    pad_h: 1
    round_mode: 1
  }
}

kewang-xlnx, 2020-10-28:
Hi, the above error showed that the number of conv1_bn_scale layer's parameters must be 2, but only 1 parameters could be found. 
According to the prototxt file, the bias_term of conv1_bn_scale layer was set 'false'. So, only the scale parameter γ in batch normalization could be found, the shift parameter β could not be found.

JerrySciangula, 2020-10-29:
Hi @kewang-xlnx, so it means that in the caffemodel file there aren't the parameters for the bias terms. Did I understand right? So, do I have retrain the network or can I fix it in another way?

closed 2021-03-11 by JerrySciangula

kewang-xlnx, 2021-05-07:
Hi @JerrySciangula, you can just change the bias_term of batch-norm layer to 'true', and run again. If it's not work, you may need to retrain the network.