Xilinx/Vitis-AI #183: [DNNDK_XRT] Cannot find device, index: 0 when using dexplorer

kevinkit, 2020-10-20:
When starting my petalinux on a ZCU106 and executing the command `dexplorer -w ` I get the following message:

`[DNNDK_XRT] Cannot find device, index: 0 `

Is this a common behaviour? Why does it not find my DPU? Is it the device-tree? Did I forgot to install something?

x3chen, 2020-10-21:
When XRT couldn't probe any device, dexplorer will report the error. 
Please review your platform first. u may find something different via commands "xbutil scan" and "cat /proc/interrupts"

kevinkit, 2020-10-21:
[image]
[image]

I am an absolute newbie when it comes to embedded Linux - but it looks like to me that the DPU is not found at all? Can I check if this is due to my Hardware Design or if I messed up the device-tree somehow?

x3chen, 2020-10-21:
ZOCL is linux kernel driver for xrt. "ZOCL: unknown" is incorrect. 
Have u added zocl node in device node? please check your project with official design.

kevinkit, 2020-10-21:
Thank you very much, I added it by using `petalinux-config -c rootfs `  and then select in the user package

[image]
Is that what is meant by device node or do I need to add it somewhere else, too? 
What is meant by the "official design" ? The TRD to include a dpu or the bsp ?

x3chen, 2020-10-21:
please use "lsmod" to check zocl has been insmod
zocl.ko is in the directory /lib/modules/`uname -r`/extra

the official platform design is as follows:
https://github.com/Xilinx/Vitis_Embedded_Platform_Source/blob/master/Xilinx_Official_Platforms/zcu104_base/petalinux/project-spec/meta-user/recipes-bsp/device-tree/files/system-user.dtsi

Tsukasa0403, 2020-10-22:
Hi, I have the same problem, but I found out that there are missing packages in the user package.
How did you create the petalinux project? Please let me know. Thank you.
 
[image]

kevinkit, 2020-10-22:
@Tsukasa0403 i do not know if these too are related, maybe open up another issue for this on how to add user-packages

qianglin-xlnx, 2022-02-22:
Hi @kevinkit 
As it has been a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please open a new issue for it. Thank you.

closed 2022-02-22 by qianglin-xlnx