Xilinx/Vitis-AI #185: Vitis AI Compiler for Caffe model error #4

JerrySciangula, 2020-10-22:
Hi everyone,

I'm a student at Scuola Superiore Sant'Anna in Pisa.

I'm trying to accelerate a network model on a ZCU102 ultrascale+.
I can successfully quantize the model using vai_q_caffe and the tool correctly generates the 4 quantize_results files.

When I try to compile the model using vai_c_caffe, it outputs:

**[VAI_C][Fatal] Check failed for condition [fixed() && fixinfo_.size() > 3] in [/home/xbuild/conda-bld/dnnc_1592904456005/work/dnnc_impl/core/quant.h:159] :Invalid fix info for [conv1], current fix info is [ 8 5 8 5 8 5 ].**

Could anyone explain me the error and how to fix it, if it is possible?

Mookel, 2020-10-23:
Hi, the error indicates that the quantization info of the 'conv1' is invalid, perhaps the quantization tool has some bugs. 
@qianglin-xlnx , who can help on this issue for the quantization?

JerrySciangula, 2020-10-23:
Oh, ok. So, I will waiting for more information

kewang-xlnx, 2020-10-28:
Hi, in quantization process, the batch normalization layer was fused to convolution layer, so conv1 layer should have 4 groups fix info (weights, bias, the scale parameter γ in batch normalization, the shift parameter β in batch normalization). The error indicates there was only 3 groups fix info. 
As the issue https://github.com/Xilinx/Vitis-AI/issues/182 said, perhaps it was also because the bias_term of Scale layer was set 'false'.

JerrySciangula, 2020-11-04:
Hi @kewang-xlnx. Ok thank you for the reply.

closed 2020-11-04 by JerrySciangula