Xilinx/Vitis-AI #186: My zcu104 board seems not find dpu

Tsukasa0403, 2020-10-23:
I am using vivado2020.1 and petalinux2020.1 on ZCU104 board. It seems like my device does not find the dpu.
Is there anything wrong with petalinux setting? or device tree?

Following is message of `dexplorer -w` ,  `xbutil scan` and `cat /proc/interrupt`
[image]
[image]
 
and following two are the device tree I have tried with petalinux

`/include/ "system-conf.dtsi"
/{
    chosen {
        bootargs = "console=ttyPS0,115200 earlyprintk root=/dev/mmcblk0p2 rw rootwait";
	  };
};

&sdhci1 {
    no-1-8-v;
};

&amba{
	dpu{
		#address-cells = <1>;
			#size-cells = <1>;
			compatible = "xilinx,dpu";
			base-addr = <0x80000000>;
	dpucore {
		compatible = "xilinx,dpucore";
		interrupt-parent = <&gic>;
		interrupts = <0x0 0x59 0x4 >;
		core-num = <0x1>;
	  };
  };  		
};
`

`/include/ "system-conf.dtsi"
/{
    chosen {
        bootargs = "console=ttyPS0,115200 earlyprintk root=/dev/mmcblk0p2 rw rootwait";
	  };
     
   dpu: dpu@80000000 {
         compatible = "deephi, dpu";
         reg = <0x0 0x80000000 0x0 0x700>;
         interrupts = <0x0 0x59 0x4>;
         interrupt-parent = <&gic>;
         core-num = <0x1>;
     };
     dpcma: dpcma {
         compatible = "deephi,cma";
     };
};

&sdhci1 {
    no-1-8-v;
};
`

kevinkit, 2020-10-23:
You can use deephi or xilinx - both are supported by the dpu driver. However, you may need to add the following into the system-conf.dtsi:

```
&amba {
	zyxclmm_drm {
		compatible = "xlnx,zocl";
		status = "okay";
		interrupt-parent = <&axi_intc_0>;
		interrupts = <0  4>, <1  4>, <2  4>, <3  4>,
			     <4  4>, <5  4>, <6  4>, <7  4>,
			     <8  4>, <9  4>, <10 4>, <11 4>,
			     <12 4>, <13 4>, <14 4>, <15 4>,
			     <16 4>, <17 4>, <18 4>, <19 4>,
			     <20 4>, <21 4>, <22 4>, <23 4>,
			     <24 4>, <25 4>, <26 4>, <27 4>,
			     <28 4>, <29 4>, <30 4>, <31 4>;
	};
};

&axi_intc_0 {
      xlnx,kind-of-intr = <0x0>;
      xlnx,num-intr-inputs = <0x20>;
      interrupt-parent = <&gic>;
      interrupts = <0 89 4>;
};

```

However that should already be in the file since it is the prebuilt into the zcu104 image

Tsukasa0403, 2020-10-24:
Excuse me. I found out that something strange when using the bsp file downloaded at 
[](https://github.com/Xilinx/Vitis-AI/blob/master/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/download_bsp.sh)
The **dpu.bb** file in ~/project-spec/meta-user/recipes-modules/dpu/
It looks as following:

------------------------------------------------------------------------------------------------
SUMMARY = "Recipe for  build an external dpu Linux kernel module"
SECTION = "PETALINUX/modules"
LICENSE = "GPLv2"
LIC_FILES_CHKSUM = "file://COPYING;md5=12f884d2ae1ff87c09e5b7ccc2c4ca7e"

inherit module

DEPENDS = "bc-native"

#dpu_driver_7aaf388f
SRC_URI = "file://Makefile \
           file://dpucore.c\
           file://dpucore.h\
           file://dpuext.c\
           file://dpuext.h\
           file://dpudef.h\
	   file://COPYING \
          "

S = "${WORKDIR}"

EXTRA_OEMAKE``` += "DPU_TARGET=1.3"
-----------------------------------------------------------------------------------------------------

I am using dpu IP DPUCZDX8G_v3_3_0. It seems version 1.4.1 but the target is 1.3 .
Is that valid or how could I get the right version of dpu file used  in petalinux? 
Thank you.

fanz-xlnx, 2020-10-27:
As the dexplorer is the legacy tool from DNNDK which may not be compatible with XRT. So the message doesn't mean DPU is not correctly deployed. I hope this tutorial work for you. Thanks.
https://github.com/Xilinx/Vitis-In-Depth-Tutorial/blob/master/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/README.md

shua1zhang, 2020-10-28:
> I am using vivado2020.1 and petalinux2020.1 on ZCU104 board. It seems like my device does not find the dpu.
> Is there anything wrong with petalinux setting? or device tree?
> 
> Following is message of `dexplorer -w` , `xbutil scan` and `cat /proc/interrupt`
> [image]
> [image]
> 
> and following two are the device tree I have tried with petalinux. Thank you.
> 
> //First
> /include/ "system-conf.dtsi"
> /{
> chosen {
> bootargs = "console=ttyPS0,115200 earlyprintk root=/dev/mmcblk0p2 rw rootwait";
> };
> };
> 
> &sdhci1 {
> no-1-8-v;
> };
> 
> &amba{
> dpu{
> #address-cells = <1>;
> #size-cells = <1>;
> compatible = "xilinx,dpu";
> base-addr = <0x80000000>;
> dpucore {
> compatible = "xilinx,dpucore";
> interrupt-parent = <&gic>;
> interrupts = <0x0 0x59 0x4 >;
> core-num = <0x1>;
> };
> };
> };
> 
> //Second
> 
> /include/ "system-conf.dtsi"
> /{
> chosen {
> bootargs = "console=ttyPS0,115200 earlyprintk root=/dev/mmcblk0p2 rw rootwait";
> };
> 
> dpu: dpu@80000000 {
> compatible = "deephi, dpu";
> reg = <0x0 0x80000000 0x0 0x700>;
> interrupts = <0x0 0x59 0x4>;
> interrupt-parent = <&gic>;
> core-num = <0x1>;
> };
> dpcma: dpcma {
> compatible = "deephi,cma";
> };
> };
> 
> &sdhci1 {
> no-1-8-v;
> };

Hi,

After generate xclbin, have you put it in the "/usr/lib"?

shua1zhang, 2020-10-28:
In the [Vitis AI user guide -Legacy DNNDK Examples ](https://www.xilinx.com/html_docs/vitis_ai/1_2/runningexamples.html#ariaid-title6 
), it says: 

"The DNNDK runtime loads DPU overlay bin from the default directory /usr/lib/. Make sure that dpu.xclbin exists under /usr/lib/ as expected before running DNNDK examples. For the downloaded ZCU102 or ZCU104 system images, dpu.xclbin is copied to /usr/lib/ by default. For the customized image, it is up to you to copy dpu.xclbin manually."

kevinkit, 2020-10-28:
> As the dexplorer is the legacy tool from DNNDK which may not be compatible with XRT. So the message doesn't mean DPU is not correctly deployed. I hope this tutorial work for you. Thanks. [...]

@fanz-xlnx is there an appropiate non-legacy tool now?

Tsukasa0403, 2020-10-28:
> In the [Vitis AI user guide -Legacy DNNDK Examples ](https://www.xilinx.com/html_docs/vitis_ai/1_2/runningexamples.html#ariaid-title6), it says:
>
> "The DNNDK runtime loads DPU overlay bin from the [...]

Thank you for the reply.
I'm using Vivado flow, so there are not dpu.xclbin generated.
And, it should work when using vitis ai library.

Tsukasa0403, 2020-10-28:
> As the dexplorer is the legacy tool from DNNDK which may not be compatible with XRT. So the message doesn't mean DPU is not correctly deployed. I hope this tutorial work for you. Thanks. [...]

Thank you for the reply.
How could I know whether I correctly deploy DPU on ZCU104 without this command?

And, when I executing compiled model, the error message is "Aborted".
Dos it means the problems are not from deploying DPU but from the way I compiled the model?

Tsukasa0403, 2020-10-28:
> You can use deephi or xilinx - both are supported by the dpu driver. However, you may need to add the following into the system-conf.dtsi:
>
> ```
> &amba {
> 	zyxclmm_drm {
> 		compatible = "xlnx,zocl"; [...]

@kevinkit
Thank you for the reply.
Is that "must" to have axi_interrupt_control in vivado or could I set the interrupt-parent of DPU and zocl both at &gic.

kevinkit, 2020-10-28:
Sadly that's out of my field, hopefully some xilinx employee can answer this - for me as well.

fanz-xlnx, 2020-10-30:
> > As the dexplorer is the legacy tool from DNNDK which may not be compatible with XRT. So the message doesn't mean DPU is not correctly deployed. I hope this tutorial work for you. Thanks.
> > https://github.com/Xilinx/Vitis-In-Depth-Tutorial/blob/master/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/README.md
> 
> @fanz-xlnx
> Thank you for the reply.
> How could I know whether I correctly deploy DPU on ZCU104 without this command?
> 
> And, when I executing compiled model, the error message is "Aborted".
> Dos it means the problems are not from deploying DPU but from the way I compiled the model?

Sorry, I may not understand your question correctly. Let me put it that way.

If it is a pure Vivado flow and you are using DNNDK as the DPU driver, the _dexplorer_ command should work. And it is not expected to have any version compatible issue, as the DNNDK is fully open sourced on Github. Please double check the DPU base address on your block design, DPU driver and device tree to make sure they are all aligned.

If you are using Vivado + Vitis AI to enable DPU, the _dexplorer_ command may not work. The debug is quite rely on what error message you get when running an application. So it is encouraged for users to switch to Vitis + Vitis AI flow for DPU integration.

kevinkit, 2020-10-30:
So to get this straight:

1. DNNDK + VIVADO -> dexplorer works, however DNNDK is legacy
2. VITIS AI + VIVADO -> dexplorer will not work, however VITIS-AI not legacy
3. VITIS + VITIS AI -> will  dexplorer work here, or is the debugging here in Vitis AI ? 

You mentioned that the DNNDK is complelty open - may I ask where to find the source code?

juannzou, 2021-01-07:
Were you able to find any solution?

I am also facing the same issue. Please see here:
https://forums.xilinx.com/t5/AI-and-Vitis-AI/DPU-TRD-for-ZCU104/m-p/1187245/highlight/true#M6256

I am actually trying to find out whether there is a way to just exchange the dpu.xclbin file, while keeping the existing board image, so that I do not have to build the image each time (the bitstream gets changed). 
Any idea in this direction?

qianglin-xlnx, 2022-02-24:
Hi @Tsukasa0403 
Has this issue been solved? If yes, could you please kindly close this issue. Thank you.

closed 2022-02-24 by Tsukasa0403