Xilinx/Vitis-AI #187: Missing dependency glog when trying to install libunilog via rpm

kevinkit, 2020-10-23:
want to install the vitis-ai runtime libraries on my ZCU16 custom device. However, upon copying the respective files and trying the following command:

rpm -ivh --force --ignoresize libtarget-factory-1.2.0-r10.aarch64.rpm

 

I get greeted with this error message:

error: Failed dependencies:
libunilog >= 1.2.0 is needed by libtarget-factory-1.2.0-r10.aarch64
/bin/sh is needed by libtarget-factory-1.2.0-r10.aarch64

qianglin-xlnx, 2020-10-26:
@kevinkit 
You can follow the step 5 in https://github.com/Xilinx/Vitis-AI/tree/v1.1/Vitis-AI-Library#setting-up-the-host to build the glog-0.4.0,  then install it on the target, as the following show.

[image]

kevinkit, 2020-10-26:
Thank you, I will try that.

Why is this mentioned in the v1.1, but not in v1.2 anymore?

kevinkit, 2020-10-26:
I tried it, but sadly the error remains:

[image]

Besides glog it also points out that /bin/sh is needed, however when I look at the folder /bin it is clearly there.

Nevetheless, I checked with `ldconfig -p | grep libglog` to show me the installed libraries:

[image]

However, when I check, libglog.so.0.4.0 is there. 
[image]

qianglin-xlnx, 2020-10-26:
@kevinkit 
add this option when install the lib: --nodeps
and try it again.

kevinkit, 2020-10-26:
thank you very much - this resulted in promising results:

[image]
I will report back when this may lead to other errors, however for now I will close this issue. 

Again, thank you very much for your help

closed 2020-10-26 by kevinkit