Xilinx/Vitis-AI #188: How to generate xclbin file, if I use vivado flow to rebuild at new hardware?

yluo39github, 2020-10-27:


davidxudong, 2020-10-28:
Hi @yluo39github 

The xclbin file cannot be generated using vivado flow.

lllkiller-jlll, 2020-11-09:
xclbin is generated by Vitis.

You have to follow these steps:
1. Build Vivado project and generate xsa file.
2. Build PetaLinux image using hw description from xsa file.
3. Create hardware platform project on Vitis, and build with sysroots from PetaLinux image.
4. Create Vitis application project with hardware platform.
5. Package dpu ip using package_xo, and include dpu.xo file to Vitis application project.
6. Build Vitis application project.
7. xclbin will be generated.

yluo39github, 2020-11-20:
> xclbin is generated by Vitis.
>
> You have to follow these steps:
>
> 1. Build Vivado project and generate xsa file.
> 2. Build PetaLinux image using hw description from xsa file.
> 3. Create hardware [...]

Thank you, I still use the Vitis to generate xclbin file, but adding my customized design in the platform file.

closed 2020-11-20 by yluo39github

dextroza, 2022-02-04:
> Thank you, I still use the Vitis to generate xclbin file, but adding my customized design in the platform file.

> xclbin is generated by Vitis.
>
> You have to follow these steps:
>
>     1. Build Vivado project and generate xsa file.
>
>     2. Build PetaLinux image using hw description from xsa file.
>
>     3. Create [...]

Thanks for the steps, but I don't know how to do 5. step. How to package dpu ip using 'package_xo' and can you share full command to do that? I don't have this command when i run from terminal :$ package_xo .... How to include dpu.xo file to Vitis application project?

Thanks,
BR, 
dextroza

dextroza, 2022-02-04:
> xclbin is generated by Vitis.
>
> You have to follow these steps:
>
>     1. Build Vivado project and generate xsa file.
>
>     2. Build PetaLinux image using hw description from xsa file.
>
>     3. Create [...]

Thanks for the steps, but I don't know how to do 5. step. How to package dpu ip using 'package_xo' and can you share full command to do that? I don't have this command when i run from terminal :$ package_xo .... How to include dpu.xo file to Vitis application project?

Thanks,
BR,
dextroza

jimheaton, 2022-02-04:
The DPU-TRD has instructions on how to build the hw in Vitis, using an existing Vitis target platform (created in steps 1-3).
https://github.com/Xilinx/Vitis-AI/tree/master/dsa/DPU-TRD