Xilinx/Vitis-AI #189: trying to call app example from vivado flow results in rcu_sched self-detected stall on CPU

kevinkit, 2020-10-28:
```
root@xilinx-zcu106-2020_1:~/app/samples/bin# env LD_LIBRARY_PATH=../lib ./resnet50 ../../img/bellpeppe-994958.JPEG
[  122.167048] rcu: INFO: rcu_sched self-detected stall on CPU
[  122.172616] rcu:     1-....: (5249 ticks this GP) idle=712/1/0x4000000000000002 softirq=1557/1557 fqs=2625
[  122.182081]  (t=5250 jiffies g=1985 q=5)
[  122.185987] Task dump for CPU 1:
[  122.189200] resnet50        R  running task        0   783    608 0x00000002
[  122.196239] Call trace:
[  122.198676]  dump_backtrace+0x0/0x140
[  122.202326]  show_stack+0x14/0x20
[  122.205627]  sched_show_task+0xf4/0x120
[  122.209453]  dump_cpu_task+0x40/0x50
[  122.213012]  rcu_dump_cpu_stacks+0xa0/0xe0
[  122.217092]  rcu_sched_clock_irq+0x52c/0x780
[  122.221347]  update_process_times+0x2c/0x68
[  122.225514]  tick_sched_handle.isra.0+0x30/0x50
[  122.230027]  tick_sched_timer+0x48/0x98
[  122.233846]  __hrtimer_run_queues+0xec/0x1e8
[  122.238099]  hrtimer_interrupt+0x110/0x2c0
[  122.242181]  arch_timer_handler_phys+0x30/0x40
[  122.246607]  handle_percpu_devid_irq+0x80/0x140
[  122.251122]  generic_handle_irq+0x24/0x38
[  122.255114]  __handle_domain_irq+0x60/0xb8
[  122.259194]  gic_handle_irq+0x5c/0xb8
[  122.262840]  el1_irq+0xb8/0x140
[  122.265967]  misc_open+0x50/0x1a0
[  122.269273]  chrdev_open+0xc4/0x200
[  122.272746]  do_dentry_open+0x104/0x398
[  122.276565]  vfs_open+0x28/0x30
[  122.279691]  path_openat+0x4a0/0x1248
[  122.283336]  do_filp_open+0x74/0xf8
[  122.286808]  do_sys_open+0x168/0x218
[  122.290367]  __arm64_sys_openat+0x20/0x28
[  122.294363]  el0_svc_common.constprop.0+0x68/0x160
[  122.299144]  el0_svc_handler+0x6c/0x88
[  122.302876]  el0_svc+0x8/0xc
```

I am using the VIVADO flow and tried the given example from the Vivado DPU_TRD, however I will get greeted by this message. I do not know how to fix this, any pointers what might be wrong?

qianglin-xlnx, 2020-10-29:
@kevinkit 
The resnet50.tar.gz example under DPU-TRD/app  is used for zcu102. If you want to run the example on other platform, you have to generate the corresponding resnet50.elf model.

kevinkit, 2020-10-29:
I did that

qianglin-xlnx, 2020-10-29:
@kevinkit 
Our engineers have looked at the issue and their suggestion is to check the interrupt configuration of the hardware and check the corresponding device tree config. It seems mismatch.

kevinkit, 2020-10-29:
I changed the core num of the DPU to 0x1 , sadly without any improvements 

`Our engineers have looked at the issue and their suggestion is to check the interrupt configuration of the hardware and check the corresponding device tree config. It seems mismatch.`

f it is a mismatch in the Interrupt pins, then the error is not only limited to the ZCU106, but for everybody that will use the provided DPU_TRD in the Vivado Flow, since the configuration comes directly from the provided tcl scripts: 

https://github.com/Xilinx/Vitis-AI/blob/master/DPU-TRD/prj/Vivado/scripts/trd_prj.tcl  and
https://github.com/Xilinx/Vitis-AI/blob/master/DPU-TRD/prj/Vivado/scripts/base/trd_bd.tcl 

The ony thing I did was changing the DPU Number from 2 to 1. However, this should not be a problem since there are two in it in the orignal design, and the system-user.dtsi has already 4 on it. 

Again thank you very much for your help and your confirmation that it may be related to the interrupts

JasonXijieJia, 2022-02-21:
Hi @kevinkit 
  How is your project going so far?  In the latest vai2.0, we have updated the interrupt connections for an easier way to auto-config the dtsi. Please have a try to see if it helps for your issues.

closed 2022-03-03 by qianglin-xlnx