Xilinx/Vitis-AI #190: Network with multiple inputs quantization error

JerrySciangula, 2020-10-28:
Hi everyone,

I'm a student at Scuola Superiore Sant'Anna in Pisa. 

I'm trying to accelerate a network model on a ZCU102 ultrascale+ using Vitis-AI tools v1.2.
The network model has multiple input tensors with shape (1, 64, 1, 80). When i try to set multiple ImageData layers for the input tensors, the vai_q_caffe tool reports: 

**_Check failed: net.has_blob(input_blob)  No blob named data found, Please rename model's input blob name as data_**

Could anyone explain me the error and how to fix it, if it is possible?

kevinkit, 2020-10-28:
the entry in the prototxt must be given the name "data"

JerrySciangula, 2020-10-29:
Hi @kevinkit, ok thank you. But what we can do if we have multiple inputs? Does the tool accept just one input?

kevinkit, 2020-10-29:
I am not quite sure, but it seems that Xilinx is working on it, see: https://forums.xilinx.com/t5/AI-and-Vitis-AI/Programming-with-Vitis-AI/td-p/1104725

They also refer to: https://github.com/Xilinx/Vitis-AI/tree/master/Vitis-AI-Library/multitask 

I do not know if this will help you, but it may be a start

JerrySciangula, 2020-10-29:
Thank you very much @kevinkit

closed 2021-03-11 by JerrySciangula

aiqiangbrave, 2022-05-23:
> > Hi everyone,
> > I'm a student at Scuola Superiore Sant'Anna in Pisa.
> > I'm trying to accelerate a network model on a ZCU102 ultrascale+ using Vitis-AI tools v1.2. The network model has multiple input tensors with shape (1, 64, 1, 80). When i try to set multiple ImageData layers for the input tensors, the vai_q_caffe tool reports:
> > **_Check failed: net.has_blob(input_blob) No blob named data found, Please rename model's input blob name as data_**
> > Could anyone explain me the error and how to fix it, if it is possible?
> 
> Did you find any solution? thanks.

hello,Do you solve this problem? thank you~~~