Xilinx/Vitis-AI #192: Vitis AI Compiler for Caffe model Warning

zhang-jinyu, 2020-10-29:
Hi,
When I t using vai_c_caffe compile the quantized model of **cf_refinedet_coco_480_360_0.96_5.08G** , it reflect me the message as follows:
**[VAI_C][Warning] Only 'channel' axis supported on DPU for Concat, current layer is [arm_priorbox].**
could you explain the warning and tell me how to fix it？

Mookel, 2020-10-29:
Hi, @zhang-jinyu 

For the DPU, only concatenation in channel axis is supported, this is why the compiler complained about it when parsing the model. But in some cases, some transformation can be leveraged by the middle end of compiler to get rid of such constraint, so this is why the compiler gives you a warning message rather than an error. If the constraint still exists after passes of middle end, the backend will output an error to indicates that the concatenation in channel axis is not supported.

Hope it helps instead of confusion,thank very much.

zhang-jinyu, 2020-11-01:
HI,@Michael Lee
I appreciate your taking the time to answer my question.According to your explanation, when compiling the quantized model in XILINX MODEL ZOO, can I ignore this warning?

Mookel, 2020-11-01:
> HI,@michael Lee
> I appreciate your taking the time to answer my question.According to your explanation, when compiling the quantized model in XILINX MODEL ZOO, can I ignore this warning?

Yes, @zhang-jinyu , you can ignore it:-).

zhang-jinyu, 2020-11-01:
@Mookel thank you very much!

closed 2020-11-01 by zhang-jinyu