Xilinx/Vitis-AI #194: Model zoo doesn't have facrec_resnet20 (Used for face recognition)

sparshgarg23, 2020-10-31:
Hello,
I see that in Library user guide,it is mentioned that you use facerec_resnet20 and facerec_resnet64  models(Caffe framework).
However,I couldn't find these models in the model zoo.Any plans of uploading these models.

qianglin-xlnx, 2020-11-02:
Hi @sparshgarg23 
These models are in Xilinx Model Zoo, as shown in the following.
[image]

You can down them from the following links. 
https://www.xilinx.com/bin/public/openDownload?filename=cf_facerec-resnet20_112_96_3.5G_1.2.zip
https://www.xilinx.com/bin/public/openDownload?filename=cf_facerec-resnet64_112_96_11G_1.2.zip

qianglin-xlnx, 2022-02-17:
Hi @sparshgarg23 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-17 by qianglin-xlnx