Xilinx/Vitis-AI #195: Vitis AI Compiler for Caffe model error #5

JerrySciangula, 2020-10-31:
Hi everyone,

I'm a student at Scuola Superiore Sant'Anna in Pisa.

I'm trying to accelerate a network model on a ZCU102 ultrascale+ using Vitis-AI tools v1.2.
The network model has multiple input tensors with shape (1, 64, 1, 80). I can successfully quantize the model, but when launching the vai_c_caffe command, it reports:

**_[VAI_C][Error] The data type of [SCNN_D_1_sum]'s inputs are not equal, input 0 has type QInt8, while input 1 has type Float32
[VAI_C][Error] Infer shape for current layer [SCNN_D_1_sum] failed._**

This layer just execute a Sum Eltwise operation between blobs.

Could anyone explain me the error and how to fix it, if it is possible?

Mookel, 2020-11-04:
Hi, @JerrySciangula 
Could you enable the '--dump' option with value 'all' during the compilation and share the quant.info file? Thanks very much.

JerrySciangula, 2020-11-04:
Hi, @Mookel. Here you have the quant.info file. It is in zip file because of the file sharing tool of Github.
[quant.zip](https://github.com/Xilinx/Vitis-AI/files/5487498/quant.zip)

Mookel, 2020-11-05:
Hi, @JerrySciangula 
I checked the quant.info and found all the inputs has not been quantized, actually I don't know why:-). But this is why the compiler complained about that.

JerrySciangula, 2020-11-05:
Thank you very much, @Mookel

closed 2020-11-05 by JerrySciangula