Xilinx/Vitis-AI #196: Alveo U50 device error on DPUv3

jimd6776, 2020-11-02:
Hi, 

After 2 or 3 successful runs on a xmodel for Alveo U50 DPU the device becomes unusable. It throws the following error. Maybe the CUs become unstable? After card reset and restart of the server the problem is solved but is there any other solution?
```
WARNING: Logging before InitGoogleLogging() is written to STDERR
F1102 01:42:13.043148   289 xrt_cu.cpp:165] Check failed: is_done cu timeout! core_idx 0  handle=0x2af9440 ENV_PARAM(XLNX_DPU_TIMEOUT) 10000 state 1 ERT_CMD_STATE_COMPLETED 4 ms 10010  bo=11is_done 0 
*** Check failure stack trace: ***
F1102 01:42:13.043248   286 xrt_cu.cpp:165] Check failed: is_done cu timeout! core_idx 1  handle=0x2af9a30 ENV_PARAM(XLNX_DPU_TIMEOUT) 10000 state 1 ERT_CMD_STATE_COMPLETED 4 ms 10010  bo=5is_done 0 
```

Thanks in advance

qianglin-xlnx, 2020-11-18:
@jimd6776 
We have to confirm whether this issue is caused by power consumption.
Do you use the model that from the Xilinx Model Zoo?  If yes, I think you can try to scale down the frequency of DPU and test it again. You can refer to the following to scale down the frequency of DPU. https://github.com/Xilinx/Vitis-AI/tree/master/alveo-hbm#dpucahx8h-overlay-frequency-scaling-down
Or improve the heat dissipation of the server.
On the other hand, you can also test whether you can run your model for a long time.

jimd6776, 2020-11-22:
The xmodel is custom and it runs in less than 1 sec in my dataset. It might crash in the 2nd or 3d run, making the device unusuable. I will scale down the frequency but the temperatures seem normal.

vennelakanti1997, 2021-01-27:
Hello . I am having the same issue when running VGG19 on Ultra96V2 (using vitis-ai-1.2.1). Have you found any solution? Thanks in advance.

qianglin-xlnx, 2022-04-11:
Hi @jimd6776 
It's been a long time for this issue. I'm going to close it. If you still have other questions, please open a new issue for it. Thank you.

closed 2022-04-11 by qianglin-xlnx