Xilinx/Vitis-AI #199: Support for ZedBoard with Vitis-AI 1.2

sl33k, 2020-11-10:
Is there any comment on the support of the ZedBoard (or any Zynq-7000 based device) with the 1.2 version of Vitis-AI?

I've seen #158, which states that it is generally possibly with some integration work.
However, the [Vitis-AI documentation](https://www.xilinx.com/html_docs/vitis_ai/1_2/gjp1592545687662.html) states that only Ultrascale+ boards are supported in custom configurations.
Finally, the [DPU part](https://www.xilinx.com/html_docs/vitis_ai/1_2/zynqdpuv31introduction.html#zhd1565044063687) of the documentation again has a subsection for Zynq7000 boards indicating that it is indeed possible to use the Vitis-AI infrastructure with any Zynq7000 based device.

Therefore I'd like some clarification on whether or not this is actually supported before I spent a lot of work trying to actually integrate it.

Thanks!
- malte

qianglin-xlnx, 2020-11-17:
hi @sl33k 
So far as I know, Vitis AI can support 32-bit system, such as Zynq7000.

lllkiller-jlll, 2020-11-20:
Hi @qianglin-xlnx .

Could you please provide any tutorial for Zynq7000?

qianglin-xlnx, 2020-11-21:
@lllkiller-jlll 
I'm sorry. I don't have this kind of tutorial yet.

closed 2022-03-03 by qianglin-xlnx

Dhanwanth1803, 2024-04-03:
@qianglin-xlnx Does Vitis AI still support `ZedBoard™ Zynq®-7000 Arm®/FPGA SoC Development Board` in the versions 3.0 and 3.5?