Xilinx/Vitis-AI #202: Using Custom Address for DPU/CPU

jalezeta, 2020-11-11:
Hey there!

I am trying to run the DPU with DNNDK using the Linux libraries and I am trying to use a custom space address for the DPU-CPU data usage ( Input tensors, Output Tensors, etc ). I have seen that every value related to physical addresses in [dpu.cpp](https://github.com/Xilinx/Vitis-AI/blob/master/mpsoc/vitis_ai_dnndk_src/n2cube/src/dpu.cpp) is limited to 32 bits. The DPU IP Product Guide Document mentions that the "reg_dpu_base_addr" is 40 bits to support an address space up to 1TB. This address is specified in the "reg_dpuX_base_addrX_l" and "reg_dpuX_base_addrX_h" registers to represent the 40 bits supported by the DPU. However, I have seen that just the lower 32 bits are passed to the DPU registers. Thus, the "reg_dpuX_base_addrX_h" is always set to 0. I suspect this is happening because the DNNDK library is using 32-bit variable types for everything related to the physical address. 

As a quick summary, I am interested in using a custom space address that cannot be represented by 32 bits, and I am trying to understand what the DNNDK library is doing for allocating its memory space. For clarification purposes, I am not interested in using the split_IO mode, which is provided as an example.

qianglin-xlnx, 2020-12-07:
Hi @jalezeta 
There is a dpu_aol_alloc_dev_mem in the dpu_xrt.cpp file, which returns dpu_aol_dev_mem_t * contains uint64_t addr_phy. Runtime code call this function to allocate device memory, and save the returned pointer, then pass it back in subsequent operations. Therefore, it can handle addresses larger than 32 bits. For the register assignment, it is implemented in the setup_regs_from_assignment function in the dpu.cpp file. This function obtains the offset and value of each register from the elf file. And a 64-bit register is divided into two 32-bit registers for processing.

qianglin-xlnx, 2022-02-20:
Hi @jalezeta 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-20 by qianglin-xlnx