Xilinx/Vitis-AI #203: VITIS v.1.3 Release Date

kevinkit, 2020-11-14:
Hello, 

In the FAQ it is stated, that:

> Vitis AI will have a new release every 3-months. Each Vitis AI release will be compatible with the most current version of Vitis tools – E.g. Vitis AI 1.0 is compatible with Vitis tools 2019.2.

Since the release of 1.2 is long time ago I would like to know when v 1.3 will come out ? 
Maybe there are some delays which cannot be foreseen, however it would be nice to know which kind of features or further support for other boards we can expect - this is crucial for further planning on supporting VITIS-AI or not. 

Thank you very much for your help

fgabel, 2020-11-16:
Also interested in this.

gkrislara, 2020-12-02:
I would like to know about the same

andyluo7, 2020-12-02:
> CAUTION: This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.

Hi Kevin,

  The FAQ needs to be updated. The release cycle will be changed to 6 month. Vitis AI 1.3 will be based on 2020.2. The release is planned to be in Dec.

BR,
Andy

kevinkit, 2020-12-02:
Hi @andyluo7 thank you very much for this information. 

Are there any other information available such as:

- Definite date of relase
- Added features
- Added "out-of-the-box" support for other boards

Greetings,

Kevin

0b11101010, 2020-12-09:
It will be released on 18th December as said on Xilinx event.

closed 2020-12-22 by andyluo7