Xilinx/Vitis-AI #204: VPL ERROR while compliling DPU-TRD

gkrislara, 2020-11-16:
Hi,
OS: Ubuntu 16.04
DPU-TRD,Vitis,Vivado,Petalinux version: 2020.1

While running
`make KERNEL=DPU DEVICE=zcu102`
VPL throws the following error

`INFO: [v++ 60-1454] Run Directory: /home/htic-broncho/DPU-TRD/prj/Vitis/binary_container_1/link/run_link

****** vpl v2020.1 (64-bit)
  **** SW Build 2902540 on Wed May 27 19:54:35 MDT 2020
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

INFO: [VPL 60-839] Read in kernel information from file '/home/htic-broncho/DPU-TRD/prj/Vitis/binary_container_1/link/int/kernel_info.dat'.
INFO: [VPL 60-423]   Target device: zcu102_dpu
INFO: [VPL 60-1032] Extracting hardware platform to /home/htic-broncho/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/.local/hw_platform
[18:35:45] Run vpl: Step create_project: Started
Creating Vivado project.
[18:35:48] Run vpl: Step create_project: Completed
[18:35:48] Run vpl: Step create_bd: Started
[18:36:00] Run vpl: Step create_bd: Failed
[18:36:00] Run vpl: FINISHED. Run Status: create_bd ERROR
ERROR: [VPL 60-773] In '/home/htic-broncho/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/runme.log', caught Tcl error:  **ERROR: [Common 17-157] Error parsing command line options, please type 'import_files -help' for usage info.**
ERROR: [VPL 60-704] Integration error, Step failed: create_bd An error stack with function names and arguments may be available in the 'vivado.log'.
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [18:36:01] Run run_link: Step vpl: Failed
Time (s): cpu = 00:00:17 ; elapsed = 00:00:43 . Memory (MB): peak = 1339.039 ; gain = 0.000 ; free physical = 905 ; free virtual = 15242
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
Makefile:70: recipe for target 'binary_container_1/dpu.xclbin' failed
make: *** [binary_container_1/dpu.xclbin] Error 1
`
Attaching vivado.log for further reference
`#-----------------------------------------------------------
# Vivado v2020.1 (64-bit)
# SW Build 2902540 on Wed May 27 19:54:35 MDT 2020
# IP Build 2902112 on Wed May 27 22:43:36 MDT 2020
# Start of session at: Mon Nov 16 18:35:33 2020
# Process ID: 29532
# Current directory: /home/htic-broncho/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl
# Command line: vivado -log vivado.log -applog -messageDb vivado.pb -mode batch -source vpl.tcl -notrace
# Log file: /home/htic-broncho/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/vivado.log
# Journal file: /home/htic-broncho/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/vivado.jou
#-----------------------------------------------------------
source vpl.tcl -notrace
INFO: Dispatch client connection id - 33609
WARNING: failed to connect to dispatch server - client already initialized
[18:35:45] Run vpl: Step create_project: Started
INFO: [OCL_UTIL] current step: vpl.create_project
Creating Vivado project.
INFO: [OCL_UTIL] internal step: create_project -part xczu9eg-ffvb1156-2-e -force prj prj
INFO: [OCL_UTIL] set_property design_mode GateLvl [current_fileset]
INFO: [Vivado 12-3498] Setting project source management mode (property SOURCE_MGMT_MODE) to "None".
INFO: [OCL_UTIL] set_property PR_FLOW 1 [current_project]
[18:35:48] Run vpl: Step create_project: Completed
[18:35:48] Run vpl: Step create_bd: Started
INFO: [OCL_UTIL] current step: vpl.create_bd
INFO: [OCL_UTIL] set_property board_part_repo_paths .local/hw_platform/board [current_project]
INFO: [OCL_UTIL] set_property board_part xilinx.com:zcu102:part0:3.3 [current_project]
INFO: [OCL_UTIL] setting ip_repo_paths: /home/htic-broncho/DPU-TRD/prj/Vitis/binary_container_1/link/int/xo/ip_repo/xilinx_com_RTLKernel_DPUCZDX8G_1_0 /home/htic-broncho/Xilinx/Vitis/2020.1/data/cache/xilin$
INFO: [OCL_UTIL] internal step: update_ip_catalog
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1700] Loaded user IP repository '/home/htic-broncho/DPU-TRD/prj/Vitis/binary_container_1/link/int/xo/ip_repo/xilinx_com_RTLKernel_DPUCZDX8G_1_0'.
INFO: [IP_Flow 19-1700] Loaded user IP repository '/home/htic-broncho/Xilinx/Vitis/2020.1/data/cache/xilinx'.
INFO: [IP_Flow 19-1700] Loaded user IP repository '/home/htic-broncho/DPU-TRD/prj/Vitis/binary_container_1/link/vivado/vpl/.local/hw_platform/ipcache'.
INFO: [IP_Flow 19-1700] Loaded user IP repository '/home/htic-broncho/Xilinx/Vitis/2020.1/data/ip'.
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/home/htic-broncho/Xilinx/Vivado/2020.1/data/ip'.
update_ip_catalog: Time (s): cpu = 00:00:08 ; elapsed = 00:00:12 . Memory (MB): peak = 2174.969 ; gain = 0.000 ; free physical = 894 ; free virtual = 15223
INFO: [OCL_UTIL] internal step: config_ip_cache -use_cache_location /home/htic-broncho/DPU-TRD/prj/Vitis/binary_container_1/ip_cache
**INFO: [OCL_UTIL] internal step: import_files -norecurse .local/hw_platform/bd/hdmi_example_zcu102.bd -of_objects
ERROR: caught error: ERROR: [Common 17-157] Error parsing command line options, please type 'import_files -help' for usage info.**

    while executing
"import_files -norecurse $hw_platform_dr_bd {*}$rm_switch"
    (procedure "ocl_util::import_dr_bd" line 40)
    invoked from within
"ocl_util::import_dr_bd $hw_platform_info $config_info"
[18:36:00] Run vpl: Step create_bd: Failed
INFO: [OCL_UTIL] current step: vpl.create_bd failed. To rerun the existing project please use --from_step vpl.create_bd
Step failed: create_bd An error stack with function names and arguments may be available in the 'vivado.log'.
INFO: [Common 17-206] Exiting Vivado at Mon Nov 16 18:36:00 2020...
`
How to rectify this issue?
Thanks in advance

qianglin-xlnx, 2020-12-07:
Hi @gkrislara 
It seems that you choose the wrong platform source. You should use ZCU102 base platform from https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-platforms/2020-1.html

qianglin-xlnx, 2022-02-20:
Hi @gkrislara 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-20 by qianglin-xlnx