Xilinx/Vitis-AI #205: How to change DPU's frequency?

yluo39github, 2020-11-20:
The default frequency for zcu104 frequency is 300MHz (CLK) and 600MHz (CLK x2). How to change them?
As I also found in the zcu104_dpu platform, there exist 7 clock source (0:150MHz, 1:300MHz, 2:75MHz. 3:100MHz, 4:200MHz, 5:400MHz, and 6:600MHz). Should I directly use them in the dpu config file?

lllkiller-jlll, 2020-11-30:
You have to change your prj_config(Link: https://github.com/Xilinx/Vitis-AI/blob/master/DPU-TRD/prj/Vitis/config_file/prj_config).

If you want DPU frequency from 300MHz to 200MHz, the example is below:
`
freqHz=200000000:DPUCZDX8G_1.aclk
freqHz=400000000:DPUCZDX8G_1.ap_clk_2
freqHz=200000000:DPUCZDX8G_2.aclk
freqHz=400000000:DPUCZDX8G_2.ap_clk_2

id=4:DPUCZDX8G_1.aclk
id=5:DPUCZDX8G_1.ap_clk_2
id=4:DPUCZDX8G_2.aclk
id=5:DPUCZDX8G_2.ap_clk_2
`

The ID is based on your platform information.

And change your dpu.xclbin  generated by Vitis on zcu104.

qianglin-xlnx, 2022-02-20:
Hi @yluo39github 
Have you solved this issue?

qianglin-xlnx, 2022-03-02:
Hi @yluo39github 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-02 by qianglin-xlnx