Xilinx/Vitis-AI #206: Post quantization input scaling / offset

qukhan, 2020-11-23:
When processing a network model to run on the DPU, can/will the quantization step apply any scaling or offset to the input nodes ?

I'm having troubles with a model that takes an image as an input (range [0; 255] when the DPU API uses `int8_t` [-128; 127]). When dumping the layers using `vai_q_tensorflow dump` it seems that all the values are divided by 2 (+1 for odd numbers).

Without applying this extra (hidden) scale in my app, the results from DPU and the dump are not the same. 

If there is such a scaling, where can I find information on that ? If not, am I doing something wrong ?

xieyi4650, 2020-12-03:
Hi @qukhan ,  the dumped value is just the result of float value that is processed by quantize and dequantize.   See
https://github.com/Xilinx/Vitis-AI/blob/master/Vitis-AI-Quantizer/vai_q_tensorflow/tensorflow/contrib/decent_q/kernels/fix_neuron_ops.h#L179    function `dp_quantize_cpu`

And when using `vai_q_tensorflow dump`,  you can set "--dump_float 1" to dump the float tensor data, and check the difference between origin input data and input data after quantize-dequantize.

qukhan, 2020-12-09:
Hi, thanks for your reply.

It seems that the `dpuGet[Input/Output]TensorScale` functions do not work properly. They always return 0.

xieyi4650, 2020-12-11:
Hi @qukhan  can you indicate where is dpuGet[Input/Output]TensorScale.  I don’t remember this function, can you tell me where the function is?

qukhan, 2020-12-14:
These are functions of the advanced API. You'll find the code of `dpuGetTensorScale`  here.

https://github.com/Xilinx/Vitis-AI/blob/master/mpsoc/vitis_ai_dnndk_src/n2cube/src/dpu.cpp#L1853-L1858

Mookel, 2020-12-16:
Hi, @qukhan 

Could you help check which version of N2Cube you are using?
We have a bug related to dpuGetTensorScale but we fixed it in Vitis-AI 1.2 version.

qianglin-xlnx, 2022-04-24:
Hi @qukhan 
It has been a long time and VAI1.2 has been released. So I close this issue. If you have any problems, please open a new issue. Thank you.

closed 2022-04-24 by qianglin-xlnx