Xilinx/Vitis-AI #207: Quantization and impact on the inputs

Laadr, 2020-11-30:
Hi,

I have trouble finding a recent document explaining how quantization is done. Is it still done using a uniform symmetric quantization ?
I am asking this question because it occurs to me that if quantization is symmetric, it becomes important to train a network with an input symmetric around zero. I am usually training my DNN with inputs rescaled between 0 and 1 and, if i am correct, by doing so, there is 1 bit that I never use when I quantize my inputs. Could you confirm ?
Thanks in advance for your help

xieyi4650, 2020-12-03:
hi @Laadr it is uniform symmetric quantization. The detailed is in https://github.com/Xilinx/Vitis-AI/blob/master/Vitis-AI-Quantizer/vai_q_tensorflow/tensorflow/contrib/decent_q/kernels/fix_neuron_ops.h

Laadr, 2020-12-03:
Thanks for your answer

closed 2020-12-03 by Laadr

xieyi4650, 2020-12-03:
@Laadr I do not know if you mean strictly symmetric like [-127, 127].  the number of  step range[-128, 127] in this quantize tool

Laadr, 2020-12-03:
Hi @xieyi4650, my point was rather to understand if it still makes sense to train a DNN with the inputs rescaled between 0 to 1. From your answer, I see that it does not and that it is better to have an input symmetric around 0 (between -1 and 1 for example).
As a complementary question, I am wondering if you plan to add options for the quantize tool or it is not your focus for now. I read for example that using more bits for the quantization of input activations and outputs activations of the network can help to improve the results. I would be interested in testing that.

xieyi4650, 2020-12-04:
Hi @Laadr ,  Yes, it is better to use symmetric input.  If you use (0,1) as input, this results half bits unused(-128, -1) like output of relu. 
We are planning to do this, but there is no exact time for implementation.