Xilinx/Vitis-AI #208: Vitis AI Compiler for Caffe model error #8

JerrySciangula, 2020-11-30:
Hi everyone,
I'm a student at Scuola Superiore Sant'Anna in Pisa.

I'm trying to accelerate a network model on a ZCU102 ultrascale+ using Vitis-AI tools v1.2.
I can successfully quantize the model, but when launching the vai_c_caffe command, it reports:

**_[VAI_C][Fatal] Check failed for condition [shape.count() == shape_.count()] in [/home/xbuild/conda-bld/dnnc_1592904456005/work/dnnc_impl/core/weights.h:161] :_**

Could anyone explain me the error and how to fix it, if it is possible?

Mookel, 2020-12-03:
Hi, @JerrySciangula 

It seems there is a shape mismatch for the weights during the reshape process, Could you share your prototxt file?
Thanks very much.

JerrySciangula, 2020-12-16:
Thank you very much @Mookel. I was wrong in something on my model. I just solved.

closed 2020-12-16 by JerrySciangula