Xilinx/Vitis-AI #209: Vitis AI Quantization pytorch version error

david132048, 2020-12-01:
I tried to run quantization by following the quick start.
I'm trying Vitis-AI-Quantizer/vai_q_pytorch/example/resnet18_quant.py on xilinx/vitis-ai-gpu:latest docker container and under vitis-ai-pytorch conda env.

I succeeded the evaluation for the float model, the command is python resnet18_quant.py --quant_mode 0, however I failed on --quant_mode 1.
This is my error. 
 [NNDCT_NOTE]: Loading NNDCT kernels...
-------- Start resnet18 test 

[NNDCT_NOTE]: Quantization calibration process start up...

[NNDCT_NOTE]: =>Parsing ResNet...
Illegal instruction (core dumped)

Does anyone know how to fix this problem?

TenzinCHW, 2020-12-01:
I believe this is the same problem as #135

david132048, 2020-12-02:
@TenzinCHW
I already knew about #135. However, I can't resolve the problem althought I tried to do in #135. I really don't know how to solve this problem.

TenzinCHW, 2020-12-03:
Based on the post by philreiter at this [link](https://forums.xilinx.com/t5/AI-and-Vitis-AI/Pytorch-Illegal-instruction-core-dumped/td-p/1134124) posted in #135 , it seems that the problem comes from pytorch. Try installing from conda with the pytorch channel instead of default channel (`conda install pytorch=1.1.0 torchvision=0.3.0 cudatoolkit=10.0 -c pytorch` works in google colab).

niuxjxlnx, 2022-02-21:
Suppose applying pytorch official channel when install pytorch and torchvision can solve all the same error. Close this issue first. Please open new issue if there is still problem.

closed 2022-02-21 by niuxjxlnx