Xilinx/Vitis-AI #210: Vitis AI PointPillars can offer models ? onnx?

zyxcambridge, 2020-12-02:
Vitis AI运行PointPillars模型，实现高精度多类别的3D目标检测

qianglin-xlnx, 2020-12-02:
@zyxcambridge 
PointPillars model is supported in VAI1.3. It will be released soon.

andyluo7, 2020-12-22:
Please refer to https://github.com/Xilinx/Vitis-AI/blob/master/models/AI-Model-Zoo/model-list/pt_pointpillars_kitti_12000_100_10.8G_1.3/model.yaml

closed 2020-12-22 by andyluo7

zyxcambridge, 2020-12-28:
> Please refer to [...]

你们的时延 FPS 包括 detection 也就是后处理部分吗，包括 nms 吗？
Does your delay FPS include detection, which is the post-processing part, including NMS?

zyxcambridge, 2020-12-28:
> 你们的时延 FPS 包括 detection 也就是后处理部分吗，包括 nms 吗？
> Does your delay FPS include detection, which is the post-processing part, including NMS?

  __TIC__(PP_total)
  __TIC__(PP_pre)
  pre_->process_net0(points);
  __TOC__(PP_pre)

  __TIC__(anchor_and_dpu)  // about 19ms, so it's shorter than the dpu0+dpu1+pp_middle time.
  thread anchor_mask_t(&PointPillarsPost::get_anchors_mask , post_.get() , pre_->pre_dict_); 
   //  anchor_mask_t.join();  // move fronter for easy debug

  thread process_net1_cleanmem_t(&PointPillarsPre::process_net1_cleanmem , pre_.get() );

  __TIC__(PP_dpu0)
  m0_.configurable_dpu_task_->run(0);
  __TOC__(PP_dpu0)

  process_net1_cleanmem_t.join();
  __TIC__(PP_middle)
  pre_->process_net1();
  __TOC__(PP_middle)

  __TIC__(PP_dpu1)
  m1_.configurable_dpu_task_->run(0);
  __TOC__(PP_dpu1)

   anchor_mask_t.join();
  __TOC__(anchor_and_dpu)

  __TIC__(PP_post)
  auto results = post_->post_process( );
  __TOC__(PP_post)
  __TOC__(PP_total)