Xilinx/Vitis-AI #211: GLIBC_2.27 not found

TenzinCHW, 2020-12-03:
I have been trying to run some of the examples from the VART samples on an Ubuntu 16.04 instance. I get an error similar to the one below regarding GLIBC_2.27 on all the examples.
```
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1201 10:01:39.340211 12394 main.cc:380] create running for subgraph: subgraph_layer0-conv(fix)
F1201 10:01:39.347483 12394 dpu_runner.cpp:155] [UNILOG][Check Failed: handle != __null][/proj/rdi/builds/vitis-ai/Vitis-AI-1.2/VART/1.2/16/runner/src/dpu_runner.cpp:155][VART_RUNNER_CONSTRUCTION_FAIL][Cannot create runner] cannot open library! lib=libvart-dpu-runner.so, error=/lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.27' not found (required by /usr/lib/libvart-dpu-runner.so)
*** Check failure stack trace: ***
Aborted
```
I have noticed from [other libraries](https://github.com/mavlink/qgroundcontrol/issues/8533) that this problem arises because Ubuntu 16 only supports up to GLIBC_2.23, and Ubuntu 18 comes with GLIBC_2.27. As the cloud service provider runs all instances as containers, I am unable to upgrade to Ubuntu 18.04 (rebooting after the upgrade causes the container to reload from the original image).

Just wondering if there is any fix for this or should I just really try to get an Ubuntu 18.04 instance?

hanxue, 2020-12-07:
Hi @TenzinCHW ,

Is there a chance you can run the Vitis AI Docker container? This will avoid the incompatible GLIBC problem you are facing on Ubuntu 16.04

Vitis AI is built on an Ubuntu 18.04 environment, and getting the libraries to run on Ubuntu 16.04 will not be easy. If you must pursue this path, there are a few options:

1. Run a `chroot`ed environment with GLIBC_2.27
2. Install (not upgrade) GLIBC_2.27 libraries in a different path, and use `patchelf` or `LD_PRELOAD` environment variable before running the VART samples

For more information, you can refer to the answers on https://stackoverflow.com/questions/847179/multiple-glibc-libraries-on-a-single-host/851229#851229

TenzinCHW, 2020-12-07:
Ah yes so the service I'm using is run on docker itself, so I just built and ran the CPU version of the Vitis AI container with Ubuntu 18. Thank you for the help @hanxue

closed 2020-12-07 by TenzinCHW