Xilinx/Vitis-AI #212: NNDCT_ERROR

david132048, 2020-12-07:
I'm trying to build pytorch_NNDCT from source. 

However, I'm getting inly one error related to pytorch_NNDCT.

I don't know how to solve this problem.

Anyone help?

This is my error log.

(test_env) root@System-Product-Name:/workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding# python -c "import pytorch_nndct"

[NNDCT_ERROR]: /workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/lib/nndct_kernels.so: undefined symbol: _ZN2at19NonVariableTypeMode10is_enabledEv

carrotandjzy, 2020-12-07:
Can you rm /workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/lib/nndct_kernels.so and try again ?

david132048, 2020-12-08:
I solved this problem.

Hongshen-Yang, 2022-01-05:
Having this issue again under 1.4.1, any suggestions?
There is no /lib under /workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn

niuxjxlnx, 2022-02-21:
Hi, @Hongshen-Yang:

In general, rebuild file /workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/lib/nndct_kernels.so can always solve the problem. If this can not solve the error you met, can you please share steps you tried to solve the problem here?

closed 2022-03-03 by qianglin-xlnx