Xilinx/Vitis-AI #213: Check failed: google::protobuf::TextFormat::Print(proto, output)

SevenKTon, 2020-12-07:
When I run the "vai_q_caffe quantize -model ResNet-50_NoBN.prototxt -weights ResNet-50_NoBN.caffemodel", I met the problem"F1206 18:07:44.795620   111 io.cpp:59   Check failed: google::protobuf::TextFormat::Print(proto, output)".I hope you would give me a hand

niuxjxlnx, 2020-12-07:
Hi, SevenKTon:

Suppose the process can not write the directory ./quantize_results. Please check if this is the root cause. If not, please send me your vitis-ai version information and your model information.

SevenKTon, 2020-12-07:
Thank you, niuxjxlnx:

I solved the problem, but I'm not quit sure if my solution is correct. I read problems from others that I find some problem about "protobuf" was caused by the version problem. But I didn't setup the protobuf for version 3.6.1 with "conda install protobuf=3.6.1" because of the read-write permission about anaconda3. So I execute "sudo chown -R 777 anaconda3". Then I execute "conda install protobuf=3.6.1" once more. But my version is still the 3.11. I use the"Resnet-50.prototxt" and "Resnet-50.caffemodel" ,and my calibration contains six figures. 
my new problem is that the loss is 0. Is my calibration too small ? I think I need a correct result to reference.
[image]
[image]
 

Thank you again for answering my questions in your busy time.

SevenKTon, 2020-12-07:
I1207 00:56:27.856424  3898 vai_q.cpp:206] Calibration iter: 95/100 ,loss: 0
I1207 00:56:30.810132  3898 vai_q.cpp:206] Calibration iter: 96/100 ,loss: 0
I1207 00:56:33.762431  3898 vai_q.cpp:206] Calibration iter: 97/100 ,loss: 0
I1207 00:56:36.756350  3898 vai_q.cpp:206] Calibration iter: 98/100 ,loss: 0
I1207 00:56:39.743417  3898 vai_q.cpp:206] Calibration iter: 99/100 ,loss: 0
I1207 00:56:42.792248  3898 vai_q.cpp:206] Calibration iter: 100/100 ,loss: 0

niuxjxlnx, 2020-12-07:
"loss" will not be maintained in calibration stage and its value does not affect calibration. 
Please check the quantized model accuracy directly and compile the quantized model.

SevenKTon, 2020-12-18:
Thank you niuxjxlnx , I'll not be entangled with the problem

closed 2020-12-21 by SevenKTon