Xilinx/Vitis-AI #214: ./docker_build_gpu.sh    returned a non-zero code: 4

Songxinlei, 2020-12-08:
when I run the command" ./docker_build_gpu.sh ",then finally  will appear the error that   "  returned a non-zero code: 4".
I don't know how to solve the error.

----------------------------------------------------------------------------------------
Step 20/31 : RUN mkdir -p /scratch && wget -q https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh -O /scratch/miniconda.sh &&         /bin/bash /scratch/miniconda.sh -b -p $VAI_ROOT/conda &&         ln -s $VAI_ROOT/conda/etc/profile.d/conda.sh /etc/profile.d/conda.sh &&         echo ". $VAI_ROOT/conda/etc/profile.d/conda.sh" >> ~/.bashrc &&         echo "conda activate base" >> ~/.bashrc
 ---> Running in ccf2eec48c43
The command '/bin/bash -c mkdir -p /scratch && wget -q https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh -O /scratch/miniconda.sh &&         /bin/bash /scratch/miniconda.sh -b -p $VAI_ROOT/conda &&         ln -s $VAI_ROOT/conda/etc/profile.d/conda.sh /etc/profile.d/conda.sh &&         echo ". $VAI_ROOT/conda/etc/profile.d/conda.sh" >> ~/.bashrc &&         echo "conda activate base" >> ~/.bashrc' returned a non-zero code: 4
---------------------------------------------------------------------------------------

hanxue, 2020-12-11:
Hi @Songxinlei ,

This problem usually is due to network issue, and failure to download Miniconda (Anaconda). Can you try the command below **outside** of Docker and see if it works?

```
wget https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh -O miniconda.sh
```

If you still experience the same error, we will need you to paste the full docker build output either here, or create a gist https://gist.github.com/

qianglin-xlnx, 2022-02-20:
Hi @Songxinlei 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-20 by qianglin-xlnx