Xilinx/Vitis-AI #216: Pytorch UNet quantization error

TenzinCHW, 2020-12-09:
I am trying to compile [this model](https://github.com/milesial/Pytorch-UNet) with the pytorch quantizer. However I am getting the following error with `quant_mode 1`.
```
Traceback (most recent call last):
  File "unet_quant.py", line 195, in <module>
    quant_mode=args.quant_mode)
  File "unet_quant.py", line 140, in quantization
    quant_mode, model, (inp))
  File "/home/t/miniconda3/envs/pv/lib/python3.6/site-packages/pytorch_nndct/apis/quant_api.py", line 92, in torch_quantizer
    quant_mode=quant_mode)
  File "/home/t/miniconda3/envs/pv/lib/python3.6/site-packages/pytorch_nndct/apis/utils.py", line 124, in prepare_quantizable_module
    graph = parse_module(module, input_args)
  File "/home/t/miniconda3/envs/pv/lib/python3.6/site-packages/pytorch_nndct/apis/utils.py", line 65, in parse_module
    module, input_args)
  File "/home/t/miniconda3/envs/pv/lib/python3.6/site-packages/pytorch_nndct/parse/parser.py", line 22, in __call__
    nndct_node.op = op_creator(self, raw_graph, raw_node)
  File "/home/t/miniconda3/envs/pv/lib/python3.6/site-packages/pytorch_nndct/parse/op_dispatcher.py", line 85, in __call__
    op = getattr(self, raw_node.kind, self.default)(*args)
  File "/home/t/miniconda3/envs/pv/lib/python3.6/site-packages/pytorch_nndct/parse/op_dispatcher.py", line 605, in constant_pad_nd
    op.set_config('pad', pad)
  File "/home/t/miniconda3/envs/pv/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_operator.py", line 261, in set_config
    setattr(self, config_name, value)
  File "/home/t/miniconda3/envs/pv/lib/python3.6/site-packages/pytorch_nndct/parse/torch_op_def.py", line 972, in pad
    self.set_attr(self.AttrName.PAD_WITH, value)
  File "/home/t/miniconda3/envs/pv/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_operator.py", line 254, in set_attr
    self._attrs[attr_name].value = attr_value
  File "/home/t/miniconda3/envs/pv/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_operator.py", line 106, in value
    f"The type of attr '{self._name.value}' should be {self._type} instead of {type(value[0])}"
TypeError: The type of attr 'pad_with' should be <class 'int'> instead of <class 'nndct_shared.nndct_graph.base_tensor.Tensor'>
```

If anyone would like to help out I can upload the quantization script and data. Any help is appreciated.

carrotandjzy, 2020-12-16:
You can check your pad operation and try to set pad with constant value instead of expression

TenzinCHW, 2020-12-25:
Oh okay thank you! I changed the value type of the inputs from `Tensor` to `int` and it successfully parsed the model. However now I am facing the error below.
```
Traceback (most recent call last):
  File "unet_quant.py", line 196, in <module>
    quant_mode=args.quant_mode)
  File "unet_quant.py", line 161, in quantization
    acc1_gen, loss_gen = evaluate(quant_model, val_loader, loss_fn)
  File "unet_quant.py", line 119, in evaluate
    mask_pred = model(images)
  File "/home/t/miniconda3/envs/pv/lib/python3.6/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "quantize_result/UNet.py", line 113, in forward
    self.output_module_1 = self.module_1(self.output_module_0)
  File "/home/t/miniconda3/envs/pv/lib/python3.6/site-packages/torch/nn/modules/module.py", line 493, in __call__
    result = self.forward(*input, **kwargs)
  File "/home/t/miniconda3/envs/pv/lib/python3.6/site-packages/pytorch_nndct/nn/modules/conv.py", line 47, in forward
    param_names=self.params_name)
  File "/home/t/miniconda3/envs/pv/lib/python3.6/site-packages/nndct_shared/quantization/utils.py", line 85, in process_inputs_and_params
    tensor_type='param')
  File "/home/t/miniconda3/envs/pv/lib/python3.6/site-packages/pytorch_nndct/quantization/torchquantizer.py", line 67, in do_scan
    bnfp[1] = (int)(Tfixpos.item())
RuntimeError: CUDA error: an illegal memory access was encountered
```
I traced the problem to an object/function called `NndctDiffsFixPos` which I think modifies `Tfixpos` depending on the value of `method` passed in. It fails when the value is 3, which based on the source means that the variable being quantized is a parameter. Further digging leads me to the use of `cuda_fix_neuron_v2` which is defined in vai_q_pytorch/include/nndct_fix_kernels.cuh`. I'm not sure exactly what's going wrong within the function itself at the moment but I'll try digging around some more. If anyone has a clue that would be great.

niuxjxlnx, 2021-01-05:
It seems that the tensor Tfixpos is not on GPU, but CPU, 
Please confirm you have set correct device for model, input data and torch_quantizer.

TenzinCHW, 2021-01-11:
I'm using the resnet quantizer script and just replaced the dataset and model, and ensured that I put the model and data on the GPU. Not sure how to put the torch_quantizer on the GPU but I checked and Tfixpos is on the GPU.

kewang-xlnx, 2021-01-13:
You can upload the quantization script and test data, I'll check the error.

TenzinCHW, 2021-01-14:
@kewang-xlnx Thank you so much. I have a feeling the problem might be how the network is defined, since it uses the `torch.nn.functional` module, which is not used in the resnet18 definition in torchvision.

Here's the link to the quantization script and test data: https://drive.google.com/file/d/1hmO2Rved9H5PUULn3rctKxRHszScm0mW/view?usp=sharing.

After unzipping and entering the directory, run the `exportweights.py` file to get the weights (weights exported from different versions of torch will not interoperate). The data directory is `data`.

kewang-xlnx, 2021-01-18:
The error is caused by the inconsistency between the cuda device of model and the input. In line 149 of unet_quant.py, the model's device is gpu:0, and in line 116 and 117, the inputs' device is gpu:1. You should make them the same.
There is another bug in the script. In line 66 of unet_quant.py, the input variable 'train' of the function is covered by the Temporary variable 'train'.

TenzinCHW, 2021-01-19:
Oh man I can't believe I missed that. Thank you so much for your help.

However now there's a problem with the model itself, which looks like something might have went wrong with converting the model because it occurs in the `UNet.py` in the `quantized_result` directory.

If I changed `diffX` and `diffY` to type `int`, running with `--quant_mode 1` runs without error but `--quant_mode 2` results in the following error when `dump_xmodel` is called:

```
F0119 11:35:47.264890  2363 shape_inference.cpp:444] [UNILOG][FATAL][XIR_INVALID_ARG_OCCUR][Invalid arg occurence!] Wrong axis! The dimensions except the axis defined by user of input feature maps of concat op should be same. But here one of the shapes is {1, 161, 238, 512}, one of the shape is {1, 160, 239, 512}, and the axis is 3
*** Check failure stack trace: ***
Aborted (core dumped)
```

However, if I leave `diffX` and `diffY` as tensors, running with `--quant_mode 1` results in the following error:

```
Traceback (most recent call last):
  File "unet_quant.py", line 197, in <module>
    quant_mode=args.quant_mode)
  File "unet_quant.py", line 162, in quantization
    acc1_gen, loss_gen = evaluate(quant_model, val_loader, loss_fn)
  File "unet_quant.py", line 119, in evaluate
    mask_pred = model(images)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "quantize_result/UNet.py", line 291, in forward
    self.output_module_77 = self.module_77(tensors=[self.output_module_27,self.output_module_76], dim=1)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/nn/modules/concat.py", line 40, in forward
    output = torch.cat(inputs, dim)
RuntimeError: invalid argument 0: Sizes of tensors must match except in dimension 1. Got 238 and 239 in dimension 3 at /opt/conda/conda-bld/pytorch_1579027003190/work/aten/src/THC/generic/THCTensorMath.cu:71
```

kewang-xlnx, 2021-01-26:
So far, to quantize UNET, the width and height of input should be multiples of 16. You can resize them to multiples of 16. I have tried to modify the code, and it work ok with **--quant_mode 1** and **--quant_mode 2**. The modifications are:
1. unet_quant.py: inp = torch.randn([batch_size, 3, 1280, 1918]) -->  inp = torch.randn([batch_size, 3, 1280, 1920
2. utils/dataset.py: newW, newH = int(scale * w), int(scale * h) --> newW, newH = math.ceil(scale * w/16)*16, math.ceil(scale * h/16)*16
3. unet/unet_part.py: comments diffY = int(diffY), diffX = int(diffX), x1 = F.pad(x1, [diffX // 2, diffX - diffX // 2, diffY // 2, diffY - diffY // 2])
The next version Vitis-AI will support  input of any size.

TenzinCHW, 2021-01-26:
Ah I see. Alright sure, thank you very much once again. I will try this and if it works, I'll close this issue. Have a great day!

eddydw, 2021-02-02:
> I am trying to compile [this model](https://github.com/milesial/Pytorch-UNet) with the pytorch quantizer. However I am getting the following error with `quant_mode 1`.
>
> ```
> Traceback (most recent [...]

Did you manage to quantize and compile your model.
I am trying to do the same?

TenzinCHW, 2021-02-28:
Hi sorry for the late reply.

Yes with the changes that @kewang-xlnx suggested, I managed to get a quantized xmodel file.

After quantizing, I used the `xcompiler` tool to get a compiled xmodel file ready for usage on FPGA as follows:
`xcompiler -i quantize_result/UNet_int.xmodel /opt/vitis_ai/compiler/arch/DPUCAHX8H/U280/arch.json -o quantize_result/UNet_compiled.xmodel`

You can use another `arch.json` for a different FPGA, just search in the `/opt/vitis_ai/compiler/arch` directory in the docker container.

You can find the files [here](https://drive.google.com/file/d/1hmO2Rved9H5PUULn3rctKxRHszScm0mW/view?usp=sharing).

closed 2021-02-28 by TenzinCHW

rahulsingh701, 2022-01-13:
> Hi sorry for the late reply.
>
> Yes with the changes that @kewang-xlnx suggested, I managed to get a quantized xmodel file.
>
> After quantizing, I used the `xcompiler` tool to get a compiled xmodel file [...]

Hey @TenzinCHW, I have been successfully able to compile the model but am facing difficulty in the inferencing code for x model. It’s a kind request that could you please share your inference code which you had written for the unet model for the Vitis Ai x model part.( where we interact with the vart library)