Xilinx/Vitis-AI #217: Extracting images data at wrong location

jasvinderkhurana, 2020-12-09:
I am following the instructions to build vitis-ai-library samples and came across that instruction to extract the data is wrong:

**Instead of :**
#cd ~
#tar -xzvf vitis_ai_library_r1.2.x_images.tar.gz -C Vitis-AI/vitis_ai_library
#tar -xzvf vitis_ai_library_r1.2.x_video.tar.gz -C Vitis-AI/vitis_ai_library

**Should be:**
 tar -zxvf vitis_ai_library_r1.2.0_images.tar.gz -C Vitis-AI/Vitis-AI-Library/overview/

Any comments?

qianglin-xlnx, 2020-12-10:
hi @jasvinderkhurana 
These steps are executed on the target, not on the host.
For the official zcu102/zcu104 images, the Vitis-AI/vitis_ai_library are pre-installed in the image and the instructions are correct.

jasvinderkhurana, 2020-12-10:
oh Ok, Thanks for the update..

closed 2020-12-10 by jasvinderkhurana