Xilinx/Vitis-AI #218: Why the format of weights is float32 in the deploy.caffemodel after quantization?

zxw602, 2020-12-10:
Hi everyone,

I am running a Resnet 50 example on Vitis-AI with ZCU104. After quantization, "deploy.caffemodel" was generated. And here is the problem:  I opened the "deploy.caffemodel" via Netron. However, it seems like the format of weights is still float 32 after quantization, instead of INT8. So, I am writing to figure out that is there any "dequantization" part during the final process of model quantization (which converts INT8 to Float 32 back again) ? Or there is just something going wrong with my code? Could anyone help me on this? Many thanks!! 

I attached the snapshot here for review.
[image]

niuxjxlnx, 2020-12-11:
The values are (INT8*quantization step) as our design, which are float values. In following deployment stage, INT8 value will be extracted from the float values. So what's you see is as expected and has no problem.

zxw602, 2020-12-18:
Thank you so much for your reply. That helps a lot 💯

closed 2020-12-18 by zxw602