Xilinx/Vitis-AI #219: How to compile quantization xmodel(in pytorch)

david132048, 2020-12-15:
I am trying to compile an example model resnet18 xmodel. I ran the below command in the Vitis-ai-user guide. 

(test_env) root@System-Product-Name:/workspace/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/nndct_shared/compile# vai_c_xir -i ../Vitis-AI-Quantizer/vai_q_pytorch/example/quantize_result/ResNet_int.xmodel -a /opt/vitis_ai/conda/pkgs/vaic-1.2.0-15_47295e60/lib/python3.6/site-packages/vaic/arch/DPUCZDX8G/ZCU104/arch.json -o ../Vitis-AI-Quantizer/vai_q_pytorch/example/quantize_result/ -n ResNet1

However, I got an error like below.

bash: vai_c_xir: command not found

If anyone knows about this issue, please give me some solution.

qianglin-xlnx, 2020-12-23:
Hi @david132048 
It seems that you are using VAI1.2. Which docker container do you use?
If possible, I recommend you use VAI1.3.

Hikaru-Furuta, 2020-12-24:
vai_c_xir -x ResNet_int.xmodel -a /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU104/arch.json -o ./ -n ResNet_int.xmodel

wanghong4compiler, 2022-02-17:
Hi @vasdvczxc ,
Does this issue still exist? Or, you've solved this issue?

qianglin-xlnx, 2022-02-25:
HI @vasdvczxc 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-25 by qianglin-xlnx