Xilinx/Vitis-AI #220: DPU-TRD: VAI_C doesn't recognize extra options dcf file

JerrySciangula, 2020-12-16:
Hi everyone,
I'm a student at Scuola Superiore Sant'Anna in Pisa.

I'm trying to accelerate a network model on a ZCU102 ultrascale+ using Vitis-AI tools v1.2.
I'm following the steps reported in [https://github.com/Xilinx/Vitis-AI/blob/master/DPU-TRD/prj/Vivado/README.md](DPU-TRD) using the pre-built version. I can successfully create the SD card and run the ResNet50 example. 

I generated the dcf file from the DPU-TRD/prj/Vivado/srcs/top/hw_handoff/top.hwh file using the "dlet" tool. When I run the vai_c_caffe command using the --options, it cannot recognize the new DPU configuration file and compiles with the standard .json file.

The following is the command I launch:

**_vai_c_caffe --prototxt deploy.prototxt --caffemodel deploy.caffemodel --arch /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU102/arch.json --output_dir ./compilation --options "{'dcf':'/workspace/dpu-12-15-2020-14-15.dcf'}"  --net_name dark_post_concat_soft_** 

How can I made the VAI_C recognize the new dcf file configuration?

lllkiller-jlll, 2020-12-17:
I think, you should make your own arch.json file.

For example,
`
{
    "target" : "DPUZDX8G",
    "dcf" : "/workspace/dpu-12-15-2020-14-15.dcf",
    "cpu" : "arm64"
}
`

And, type command like below:

**_vai_c_caffe --prototxt deploy.prototxt --caffemodel deploy.caffemodel --arch <your_arch_path>.json --output_dir ./compilation --net_name dark_post_concat_soft_**

I hope it is helpful for you.

JerrySciangula, 2020-12-17:
Thank you for your reply! I just tried your solution, but it doesn't work. The custom DPU should have the softmax core inside, so what I expect from the VAI_C is not: 

**_[VAI_C][Warning] layer [softmax] (type: Softmax) is not supported in DPU, deploy it in CPU instead._**

lllkiller-jlll, 2020-12-18:
Hi @JerrySciangula .

Did you check properties of **_dpu_dark_post_concat_soft_0.elf_** file with ddump?
If your dcf is not applied, your elf file is not matched with your DPU configurations.
But if your elf file's properties are matched with your DPU configurations, it means that your dcf was applied.

I also saw that some model's softmax is not used with Xilinx Softmax IP.
Which is your defined softmax layer's type?

JerrySciangula, 2021-02-11:
Hi @lllkiller-jlll,
what do you mean with your last question?

qianglin-xlnx, 2022-03-10:
Hi @JerrySciangula 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-10 by qianglin-xlnx