Xilinx/Vitis-AI #221: DPU-TRD: Vitis-AI Softmax IP API

JerrySciangula, 2020-12-17:
Hi everyone,
I'm a student at Scuola Superiore Sant'Anna in Pisa.

I'm trying to accelerate a network model on a ZCU102 ultrascale+ using Vitis-AI tools v1.2.
I'm following the steps reported in https://github.com/Xilinx/Vitis-AI/blob/master/DPU-TRD/prj/Vivado/README.md using the pre-built version. I can successfully create the SD card and run the ResNet50 example.

I generated the dcf file from the DPU-TRD/prj/Vivado/srcs/top/hw_handoff/top.hwh file using the "dlet" tool. 

For my network, I would like to use the Softmax IP core. Are there any Vitis-AI APIs to use it? 

I know there exist something in DNNDK. Is there something similar that enable me to use softmax in Vitis-AI?

lllkiller-jlll, 2020-12-18:
Hi @JerrySciangula .

Vitis AI API supports **_dpuRunSoftmax()_**.
Please refer UG1414.

[image]

JerrySciangula, 2020-12-18:
Hi @lllkiller-jlll, 

I already tried that API, but it when I used it with the dnndk samples, an error is rised up: 

**_[DNNDK XRT] Cannot find device, index: 0_**

It seems like it cannot find the DPU with that example, but if I try some Vitis-AI examples it successfully find the DPU and ends up without any error.

lllkiller-jlll, 2020-12-21:
Hi @JerrySciangula .

I think, there is a mismatching in your workflow.
Your error said that you are using XRT. Right?
But You said that you built your hw design in Vivado project.

XRT is a driver for Vitis kernels. If you want using XRT, You have to use DPU as kernel. But you did not.

If you want to build your hardware on Vivado tool, You should use DPU driver, not XRT, and you have to configure your device tree on PetaLinux.

JerrySciangula, 2020-12-21:
Hi @lllkiller-jlll, thank you very much. Is there some user guide or other documents about this argument?

lllkiller-jlll, 2020-12-22:
Hi @JerrySciangula .

Unfortunately, I cannot find detailed documents.
Anyway, I'll attach a part of UG1414.

[image]

closed 2021-03-11 by JerrySciangula