Xilinx/Vitis-AI #223: The loss of calibration iteration always equals to zero

zxw602, 2020-12-18:
Hi guys :),

I am running vai_q_caffe to generate a quantized model by the line "vai_q_caffe quantize -model float.prototxt -weights float.caffemodel". And the command executed successfully without any bugs. However, I noticed that the loss of calibration iteration always equals to zero during the process. Here is an example below. Could anyone give me any clue about this? What should I do to pinpoint the cause of the problem? Many thanks!

[image]

Songxinlei, 2020-12-18:
I got the same problem .And I got other error when I deploy the prototxt and caffe. That is the same layer name in the prototxt file.Do you have the problem?

niuxjxlnx, 2020-12-21:
"loss" will not be maintained in calibration stage and its value does not affect calibration.
Please check the quantized model accuracy directly and compile the quantized model.

qianglin-xlnx, 2022-02-20:
Hi @zxw602 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-20 by qianglin-xlnx