Xilinx/Vitis-AI #224: Error building docker image of CPU in Vitis AI v1.3.

lllkiller-jlll, 2020-12-18:
Hi.

I was trying to use Vitis AI v1.3 released.
But I'm faced with problem to build docker image of CPU.
The error message is below:

 ---> Using cache
 ---> 4fdb8b2c11ab
Step 28/46 : RUN cd /tmp     && wget --progress=dot:mega http://xcogpuvai01:8000/sdk-1013.sh     && chmod a+x /tmp/sdk-1013.sh     && /tmp/sdk-1013.sh -y     && rm -fr /tmp/*     && chmod 777 /opt/petalinux/2020.2/sysroots/aarch64-xilinx-linux/
 ---> Running in d0937abf4273
--2020-12-17 19:26:54--  http://xcogpuvai01:8000/sdk-1013.sh
Resolving xcogpuvai01 (xcogpuvai01)... failed: Temporary failure in name resolution.
wget: unable to resolve host address ‘xcogpuvai01’
The command '/bin/bash -c cd /tmp     && wget --progress=dot:mega http://xcogpuvai01:8000/sdk-1013.sh     && chmod a+x /tmp/sdk-1013.sh     && /tmp/sdk-1013.sh -y     && rm -fr /tmp/*     && chmod 777 /opt/petalinux/2020.2/sysroots/aarch64-xilinx-linux/' returned a non-zero code: 4

How can I fix it?

andyluo7, 2020-12-18:
@lllkiller-jlll , this issue was replicated. It will be fixed soon.

nhphuong91, 2021-08-19:
Has this issue been fixed? I tried to find prebuilt image on dockerhub but they are up to version 1.3.4xx or 1.3.5xx. Can't find a way to get Vitis AI (cpu) v1.3 image.