Xilinx/Vitis-AI #225: Error building docker gpu version

david132048, 2020-12-18:
I'm trying to build gpu_docker. However, I got some error.. 

g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
g++: internal compiler error: Killed (program cc1plus)
Source/CMakeFiles/CMakeLib.dir/build.make:2961: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmInstallFilesCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:1206: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmGeneratorTarget.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:3052: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmLoadCommandCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:3039: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmLoadCacheCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:2636: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmFindCommon.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:3689: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmVariableWatchCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:3494: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmTargetCompileOptionsCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:3533: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmTargetLinkDirectoriesCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:2532: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmExecProgramCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:3325: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmSetSourceFilesPropertiesCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:3078: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmMakeDirectoryCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:452: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmComputeLinkDepends.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:933: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmFileAPICMakeFiles.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:2792: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmGetPropertyCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:2935: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmInstallCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:3442: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmSubdirCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:2610: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmFileCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:2155: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmake.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:959: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmFileInstaller.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:1908: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmState.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:3637: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmUnsetCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:1869: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmSourceFile.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:3130: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmOptionCommand.cxx.o' failed
g++: internal compiler error: Killed (program cc1plus)
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [Source/CMakeFiles/CMakeLib.dir/cmInstallFilesCommand.cxx.o] Error 4
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
g++: internal compiler error: Killed (program cc1plus)
Source/CMakeFiles/CMakeLib.dir/build.make:3676: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmVariableRequiresCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:3182: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmProjectCommand.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:3793: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmGlobalGhsMultiGenerator.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:2623: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmFindBase.cxx.o' failed
Source/CMakeFiles/CMakeLib.dir/build.make:2467: recipe for target 'Source/CMakeFiles/CMakeLib.dir/cmCoreTryCompile.cxx.o' failed
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [Source/CMakeFiles/CMakeLib.dir/cmGeneratorTarget.cxx.o] Error 4
make[2]: *** [Source/CMakeFiles/CMakeLib.dir/cmLoadCommandCommand.cxx.o] Error 4
make[2]: *** [Source/CMakeFiles/CMakeLib.dir/cmLoadCacheCommand.cxx.o] Error 4
make[2]: *** [Source/CMakeFiles/CMakeLib.dir/cmFindCommon.cxx.o] Error 4
make[2]: *** [Source/CMakeFiles/CMakeLib.dir/cmVariableWatchCommand.cxx.o] Error 4
make[2]: *** [Source/CMakeFiles/CMakeLib.dir/cmTargetCompileOptionsCommand.cxx.o] Error 4
[... 245 lines cut ...]
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [Source/CMakeFiles/CMakeLib.dir/cmInstallTargetGenerator.cxx.o] Error 4
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [Source/CMakeFiles/CMakeLib.dir/cmFindLibraryCommand.cxx.o] Error 4
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [Source/CMakeFiles/CMakeLib.dir/cmPolicies.cxx.o] Error 4
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [Source/CMakeFiles/CMakeLib.dir/cmExtraEclipseCDT4Generator.cxx.o] Error 4
make[1]: *** [Source/CMakeFiles/CMakeLib.dir/all] Error 2
CMakeFiles/Makefile2:2184: recipe for target 'Source/CMakeFiles/CMakeLib.dir/all' failed
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
The command '/bin/bash -c cd /tmp; wget --progress=dot:mega https://github.com/Kitware/CMake/releases/download/v3.16.1/cmake-3.16.1.tar.gz     && tar xvf cmake-3.16.1.tar.gz     && cd cmake-3.16.1     && ./configure     && make -j     && make install     && rm -fr /tmp/*     && ldconfig' returned a non-zero code: 2

How can I fix it?

giuseppeCoccia, 2020-12-18:
Same error also for me. Is the `cmake 3.16.1` build broken?

andyluo7, 2020-12-18:
you may run out of system memory. https://github.com/kaldi-asr/kaldi/issues/3781

hanxue, 2021-03-17:
Hi @vasdvczxc @giuseppeCoccia ,

We have not encountered this issue before. And like @andyluo7 said, this could be a resource constraint / hardware issue. 

The latest version of Vitis AI 1.3.1 no longer builds cmake, and installs from a deb package. Can you try to clone this repository again and build the GPU docker image?

https://github.com/Xilinx/Vitis-AI/blob/master/setup/docker/docker/DockerfileGPU

giuseppeCoccia, 2021-03-17:
Hi @hanxue , sorry for not updating this issue.
As you said, it was a memory problem and I was able to build it on another machine with more memory available.
Thanks for your help.

hanxue, 2021-03-17:
Hi @giuseppeCoccia , glad to know the cause of the error, and was resolved.

closed 2021-03-17 by andyluo7

hanxue, 2021-09-07:
Hi @giuseppeCoccia 

I know that the issue has been resolved, I got another report of a similar issue about docker build failed without sufficient memory. Any chance you recall the amount of memory that is insufficient, and how much memory does the machine with successful build has?

rpitonak, 2022-01-27:
Hello @hanxue,

In my case machine with `16Gb` RAM (`g4dn.xlarge` EC2 machine) is not sufficient and will run out of RAM at this step:
https://github.com/Xilinx/Vitis-AI/blob/d02dcb6041663dbc7ecbc0c6af9fafa087a789de/setup/docker/dockerfiles/vitis-ai-gpu.Dockerfile#L151-L161

As well as this step needs more than `20Gb`:
https://github.com/Xilinx/Vitis-AI/blob/d02dcb6041663dbc7ecbc0c6af9fafa087a789de/setup/docker/dockerfiles/vitis-ai-gpu.Dockerfile#L162-L174

Build on the machine with `32Gb` is working fine.