Xilinx/Vitis-AI #226: vitis-AI quantize Caffe  Error

Songxinlei, 2020-12-18:
Hello, I quantized the caffe prototxt and caffemodel.

but there are some errors when I  try to deploy.
`vai_q_caffe quantize
 -model test.prototxt
 -weights trainval.caffemodel
`
`vai_c_caffe 
 --prototxt quantize_results/deploy.prototxt
 --caffemodel quantize_results/deploy.caffemodel
 --arch /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU102/arch.json 
--output_dir model 
--net_name hasco_7
`
`**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[VAI_C][Error] There are multiple layers with same name existed in deploy.prototxt, the duplicate name is [input].
[VAI_C][Error] Parsing layer graph failed.
`
I don't know why got the same layer name.
`layer {
  name: "input"
  type: "Input"
  top: "data"
  transform_param {
  }
  input_param {
    shape {
      dim: 1
      dim: 3
      dim: 352
      dim: 640
    }
  }
}
layer {
  name: "input"
  type: "Input"
  top: "data"
  phase: TRAIN
  input_param {
    shape {
      dim: 1
      dim: 3
      dim: 352
      dim: 640
    }
  }
}`

lllkiller-jlll, 2020-12-21:
Remove 

_layer {
name: "data"
type: "ImageData"
top: "data"
top: "label"
include {
phase: TRAIN
}
transform_param {
mirror: false
crop_size: 224
mean_value: 104
mean_value: 107
mean_value: 123
}
image_data_param {
source: "./../calibration_1ch.txt"
root_folder: "./../image/"
batch_size: 1
shuffle: false
}
input_param {
shape {
dim: 1
dim: 1
dim: 224
dim: 224
}
}
}_

And re-compile your quantized model.

Songxinlei, 2020-12-21:
I found that this error was caused by not adding calibration data, and it has been resolved now, thank you! @lllkiller-jlll

closed 2020-12-21 by Songxinlei

jebreca, 2022-02-23:
> I found that this error was caused by not adding calibration data, and it has been resolved now, thank you! @lllkiller-jlll

Hey @Songxinlei, I am facing the same issue, what do you mean with "by not adding calibration data", where did you add it to solve the problem?

Thanks in advance,

Jen