Xilinx/Vitis-AI #227: Vitis-AI quantize caffe model error

SevenKTon, 2020-12-18:
Hello, I quantize my prototxt and caffemodel ,but I meet some error as the following figure.
[image]
The first two layers of prototxt is as followed:
name: "model"
layer {
  name: "data"
  type: "ImageData"
  top: "data"
  top: "label"
  include {
    phase: TRAIN
  }
  transform_param {
    mirror: false
    crop_size: 224
    mean_value: 104
    mean_value: 107
    mean_value: 123
  }
  image_data_param {
    source: "./../calibration_1ch.txt"
    root_folder: "./../image/"
    batch_size: 1
    shuffle: false
  }
  input_param {
    shape {
      dim: 1
      dim: 1
      dim: 224
      dim: 224
    }
  }
}
layer {
  name: "conv1"
  type: "Convolution"
  bottom: "data"
  top: "conv1"
  convolution_param {
    num_output: 64
    bias_term: true
    pad: 1
    kernel_size: 3
    stride: 1
    weight_filler {
      type: "xavier"
    }
    bias_filler {
      type: "constant"
    }
    dilation: 1
  }
}

If anyone has some ideas, I will be very grateful.

lllkiller-jlll, 2020-12-21:
The error said that your caffemodel file and prototxt file are not match.

Please try to modify 

_layer {
name: "data"
type: "ImageData"
top: "data"
top: "label"
include {
phase: TRAIN
}
transform_param {
mirror: false
crop_size: 224
mean_value: 104
mean_value: 107
mean_value: 123
}
image_data_param {
source: "./../calibration_1ch.txt"
root_folder: "./../image/"
batch_size: 1
shuffle: false
}
input_param {
shape {
dim: 1
dim: 1
dim: 224
dim: 224
}
}
}_

to 

_layer {
name: "data"
type: "ImageData"
top: "data"
top: "label"
include {
phase: TRAIN
}
transform_param {
mirror: false
crop_size: 224
mean_value: 104
mean_value: 107
mean_value: 123
}
image_data_param {
source: "./../calibration_1ch.txt"
root_folder: "./../image/"
batch_size: 1
shuffle: false
}
input_param {
shape {
dim: 1
**dim: 1**
dim: 224
dim: 224
}
}
}_

SevenKTon, 2020-12-21:
hello lllkiller-jlll:
Thank you for your help. I have tried the method you told me, but the mistake is still existing. 
[image]
Once I used a prototxt without input data mistakenly, the program seems run normally. I think if the quantization can not dispose of one channel net such as my net.  May I make so bold as to ask whether you could try a net with one channel?
I'm very sorry that I can't send my original prototxt to you, our regulations don't allow.

SevenKTon, 2020-12-21:
I have solved the problem, I changed an 3 channels model, both prototxt and caffemodel. The two net I have met the problem runs correctly after I rewrite the caffemodel from 1 channel to 3 channels.

closed 2020-12-21 by SevenKTon