Xilinx/Vitis-AI #228: why the latest version of vitis-ai is not as fast as DNNDK?

Songxinlei, 2020-12-19:
I feel that DNNDK runs a single picture faster than DPU runs a single picture。
I feel that the time for DNN to run a single image is 0.21s, and the time to run 2000 images is 185.014s, but I use vitis-ai v1.2, the time to run a single image is 0.28s, and the time to run 2000 images It is 460.538 s. I want to know why the latest version of vitis-ai is not as fast as DNNDK?

qianglin-xlnx, 2020-12-22:
Hi @Songxinlei 
Which model did you test? Do you use multi-threading?

qianglin-xlnx, 2022-02-17:
Hi @Songxinlei 
Since we haven't received your reply for a long time,  I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-17 by qianglin-xlnx