Xilinx/Vitis-AI #229: Error on quantization pytorch model

david132048, 2020-12-21:
I'm trying to quantize pytorch example model. However, I got an error. 

(vitis-ai-pytorch) yongwon@System-Product-Name:/workspace/tools/Vitis-AI-Quantizer/vai_q_pytorc
h/example$ python resnet18_quant.py --quant_mode calib --subset_len 200

[NNDCT_NOTE]: Loading NNDCT kernels...
-------- Start resnet18 test 
Traceback (most recent call last):
  File "resnet18_quant.py", line 283, in <module>
    file_path=file_path)
  File "resnet18_quant.py", line 212, in quantization
    quant_mode, model, (input), device=device)
TypeError: torch_quantizer() got an unexpected keyword argument 'device'

If anyone knows about this issue, please give me some solution.

carrotandjzy, 2020-12-21:
Can you show me the message of print (pytorch_nndct.__version__) ?

carrotandjzy, 2020-12-21:
Can you tell me your vitis-ai version ?

david132048, 2020-12-21:
My version is 1.3.

I resolve it. 

Thank you.