Xilinx/Vitis-AI #231: ERROR: NO FRONT END SPECIFIED : compile error with vai_c_xir using PyTorch

Hikaru-Furuta, 2020-12-22:
Issue~
Thank you for the recent update.
Some issues I had with the previous version, 1.2, were solved in the new version.
But unfortunately, I still have some problems as I try to compile models.

Though I was able to quantize a ResNet18 model, which is described in the user guide, and had created essensial files for compiling, I suffered an error with compiling these files as follows.

I don't know what to do for addressing this error.
Especially I woukd like to know how to set '-n' in
'vai_c_xir -x ResNet_int.xmodel -a quant_info.json -o ./ -n ResNet_int2.xmodel'
Any advices would be great to me.

##Quantizing a model
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

(vitis-ai-pytorch)example > 
python resnet18_quant.py --data_dir ./small --model_dir . --subset_len 200 --quant_mode test --deploy resnet.xmodel

[NNDCT_NOTE]: Loading NNDCT kernels...
-------- Start resnet18 test 
Warning: Exporting xmodel needs batch size to be 1 and only 1 iteration of inference, change them automatically!
[NNDCT_NOTE]: Quantization test process start up...
[NNDCT_NOTE]: =>Quant Module is in 'cpu'.
[NNDCT_NOTE]: =>Parsing ResNet...
[NNDCT_NOTE]: =>Doing weights equalization...
[NNDCT_NOTE]: =>Quantizable module is generated.(quantize_result/ResNet.py)
[NNDCT_NOTE]: =>Get module with quantization.
100%|███████████████████████████████████████████████████████| 1/1 [00:00<00:00,  6.51it/s]
loss: 11.3088
top-1 / top-5 accuracy: 0 / 0

[NNDCT_NOTE]: =>Converting to xmodel ...

[NNDCT_NOTE]: =>Successfully convert 'ResNet' to xmodel.(quantize_result/ResNet_int.xmodel)
-------- End of resnet18 test 

#I successfully obtained quantize_result, 
which includes '__pycache__', 'bias_corr.pth', 'quant_info.json', 'ResNet.py' and 'Resnet_int.xmodel'.
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

##Compiling model files into onnx-file
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

(vitis-ai-pytorch) example/quantize_result >
 vai_c_xir -x ResNet_int.xmodel -a quant_info.json -o ./ -n ResNet_int2.xmodel
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
ERROR: NO FRONT END SPECIFIED

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Hikaru-Furuta, 2020-12-24:
I've solved myself. `quant_info.json` was not meant to be used for this.
Compiler just needs `/opt/vitis_ai/compiler/arch./arch.json` instead.

closed 2020-12-24 by Hikaru-Furuta

Yiheng95, 2020-12-29:
Thanks for sharing your solution. I also encountered the same problem when compiling pytorch ResNet18 .xmodel. 

But I cannot find the right arch.json in my directories. I am using vitis-ai built from docker recipe and the only arch.json is under Vitis-ai/demo/Whole-App-Acceleration/yolo/arch.json. Using that does not solve the "NO FRONT END SPECIFIED" error. 

Could you please share the content of your arch.json? Thanks

Hikaru-Furuta, 2021-01-07:
> .

Mine is this
~~~~~~~~~~~~~~~~~~~
{
   "target": "DPUCADX8G",
   "kernel": "xdnn",
   "lib": "{LIBXDNN_PATH}",
   "xclbin": "/opt/xilinx/overlaybins/xdnnv3",
   "weights": "weights.h5",
   "compiler": "compiler.json",
   "quantizer": "quantizer.json",
   "filename": "",
   "config_file": ""
}

~~~~~~~~~~~~~~~~~~~

Yiheng95, 2021-01-14:
Thank you! I have found the correct arch.sjon. For me its in the docker environment /opt/vitis_ai/compiler/arch/DPUCAHX8H/U50/arch.json

Hikaru-Furuta, 2021-02-08:
dfpJhICK85VR1TykhCnt