Xilinx/Vitis-AI #234: Is DPU-for-RNN supported for Alveo U50?

lllkiller-jlll, 2020-12-22:
Hi.

I read DPU-for-RNN in (https://github.com/Xilinx/Vitis-AI/tree/master/dsa/DPU-for-RNN).
In this document, DPU for RNN is supported for U25 and U50LV. But document of RNN tool usage(https://github.com/Xilinx/Vitis-AI/tree/master/tools/RNN) said that RNN compiler generate the instructions for U25 and U50.
It makes me confused. Is DPU for RNN supported for Alveo U50? Or just supported for U50LV?

qianglin-xlnx, 2020-12-22:
Hi @lllkiller-jlll 
Thank you for your feedback. It's a typo in https://github.com/Xilinx/Vitis-AI/blob/master/tools/RNN/README.md.
RNN supports U25 and U50LV.

closed 2020-12-22 by andyluo7

lllkiller-jlll, 2020-12-28:
Hi @qianglin-xlnx .

Could you tell me why DPU-RNN is supported only U50LV, not U50?