Xilinx/Vitis-AI #236: pycaffe caffe.set_device  will occupy gpu memory,and use cpu to inference.

York1996OutLook, 2020-12-22:
caffe.set_mode_gpu()    # 用了这个才是真的在gpu上推理
        caffe.set_device(gpu)   # 设置用哪个gpu

qianglin-xlnx, 2020-12-24:
Hi @York1996OutLook 
Could you describe the issue you encountered? Which model and which code? Thank you.

York1996OutLook, 2020-12-24:
in caffe\cf_FPN-resnet18_Endov_240_320_13.75G_1.3\code\test\test.py
113    caffe.set_device(int(args.gpu))

above code line didnt work well. 
and the inference is execute on the cpu,so its very slow, you should add 'caffe.set_mode_gpu()' before ' caffe.set_device(int(args.gpu))'.

York1996OutLook, 2020-12-24:
and in your examples,some code is has both of them, some not.
it was confused

wanglixilinx, 2020-12-24:
hi, @York1996OutLook , thx for your comments.  Sorry for my  careless.  I will fix it later.

wangxd-xlnx, 2022-02-22:
Fixed. Thanks all.

closed 2022-02-22 by qianglin-xlnx