Xilinx/Vitis-AI #237: Segmentation fault while compiling pytorch models with Alveo-U250

Hikaru-Furuta, 2020-12-23:
~issue~
I'm trying to compile pytorch models using vai_c_xir in vitis-ai-pytorch.
Compile code, vai_c_xir -x quantize_result/ResNet_int.xmodel -a arch.json --net_name NETNAME -o OUTPUTPATH,
doesn't seem to work correctly. Although it works fine halfway through, makes an error then.

I had Set DPU to DPUCADF8H/U250 like follows.

(vitis-ai-pytorch) Vitis-AI /workspace/tools/Vitis-AI-Quantizer/vai_q_pytorch/example >
vai_c_xir -x quantize_result/ResNet_int.xmodel -a /opt/vitis_ai/compiler/arch/DPUCADF8H/U250/arch.json --net_name net_name -o compile_output

Segmentation fault occered after 'Success boost'

**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20201221-235436-157"
[UNILOG][INFO] Compiling subgraph : subgraph_ResNet__ResNet_AdaptiveAvgPool2d_avgpool__483
 ############################################# 
 ######  Hyper Graph Construction 
 ############################################# 
 ############################################# 
 ######  Hyper Graph Construction
 ############################################# 
Floyd & Warshall
BFS
 ############################################# 
 ######  Parameters Assimilation: # 62 
 ############################################# 
 ############################################# 
 ######  Assimilating Fix Neurons: # 30 
 ############################################# 
 ############################################# 
 ######  Assimilating Relu: # 17 
 ############################################# 
 ############################################# 
 ######  Assimilating LeakyRelu: # 0 
 ############################################# 
 ############################################# 
 ######  I like VALID more than SAME
 ############################################# 
 ############################################# 
 ######  I like VALID more than SAME: # 0 
 ############################################# 
 ############################################# 
 ######  Assimilating Padding:# 0 
 ############################################# 
 ############################################# 
 ######  CPU nodes Must Go
 ############################################# 
Inputs ['ResNet__input_0_fix']
Outputs ['ResNet__ResNet_AdaptiveAvgPool2d_avgpool__483']
FPGA True: data       ResNet__input_0_fix  
FPGA True: conv2d     ResNet__ResNet_Conv2d_conv1__input_2  
FPGA True: maxpool2d  ResNet__ResNet_MaxPool2d_maxpool__input_4  
FPGA True: conv2d     ResNet__ResNet_Sequential_layer1__BasicBlock_0__Conv2d_conv1__input_5  
FPGA True: conv2d     ResNet__ResNet_Sequential_layer1__BasicBlock_0__Conv2d_conv2__input_8  
FPGA True: add        ResNet__ResNet_Sequential_layer1__BasicBlock_0__input_9  
FPGA True: conv2d     ResNet__ResNet_Sequential_layer1__BasicBlock_1__Conv2d_conv1__input_11  
FPGA True: conv2d     ResNet__ResNet_Sequential_layer1__BasicBlock_1__Conv2d_conv2__input_14  
FPGA True: add        ResNet__ResNet_Sequential_layer1__BasicBlock_1__input_15  
FPGA True: conv2d     ResNet__ResNet_Sequential_layer2__BasicBlock_0__Conv2d_conv1__input_17  
FPGA True: conv2d     ResNet__ResNet_Sequential_layer2__BasicBlock_0__Conv2d_conv2__input_20  
FPGA True: conv2d     ResNet__ResNet_Sequential_layer2__BasicBlock_0__Sequential_downsample__Conv2d_0__input_21  
FPGA True: add        ResNet__ResNet_Sequential_layer2__BasicBlock_0__input_22  
FPGA True: conv2d     ResNet__ResNet_Sequential_layer2__BasicBlock_1__Conv2d_conv1__input_24  
FPGA True: conv2d     ResNet__ResNet_Sequential_layer2__BasicBlock_1__Conv2d_conv2__input_27  
FPGA True: add        ResNet__ResNet_Sequential_layer2__BasicBlock_1__input_28  
FPGA True: conv2d     ResNet__ResNet_Sequential_layer3__BasicBlock_0__Conv2d_conv1__input_30  
FPGA True: conv2d     ResNet__ResNet_Sequential_layer3__BasicBlock_0__Conv2d_conv2__input_33  
FPGA True: conv2d     ResNet__ResNet_Sequential_layer3__BasicBlock_0__Sequential_downsample__Conv2d_0__input_34  
[... 315 lines cut ...]
 ######  Success boost
 ############################################# 
 BUFFER SIZE and RELATIVE ADDRESS 
 PARAMETER  BUFFER SIZE 0xabe800 and RELATIVE ADDRESS 0x0
 INPUT      BUFFER SIZE 0xc4000 and RELATIVE ADDRESS 0xabe800
 OUTPUT BUFFER SIZE 0x800 and RELATIVE ADDRESS 0xb82800
 SWAP BUFFER SIZE 0x0 and RELATIVE ADDRESS 0xb83000
 INSTRUCTION BUFFER SIZE 0xabe080 and RELATIVE ADDRESS 0x30000000
 BUFFER SIZE and RELATIVE ADDRESS 

    REG_ID_FORMATTED = 0
    REG_RESULT_SPACE = 1
    REG_SWAP_SPACE   = 2
    REG_PAR_SPACE    = 3
    
    ADD_REG_ID_FORMATTED = 0x10000000
    ADD_REG_RESULT_SPACE = 0x10000000
    ADD_REG_SWAP_SPACE   = 0x10000000
    ADD_REG_PAR_SPACE    = 0x10000000

    druDstBufSize: 16777216,
    
{}
[UNILOG][INFO] Compiling subgraph : subgraph_ResNet__ResNet_Linear_fc__490
Segmentation fault (core dumped)

ayeffkay, 2020-12-26:
As I understand from the [docs,](https://www.xilinx.com/html_docs/vitis_ai/1_2/izm1583734065322.html) `vai_c_xir` works correctly only for `DPUCAHX8H`. I tried to make compilation with other DPUs, but all attempts failed.

Hikaru-Furuta, 2021-01-06:
Thanks.
What's your Vitis version? I tried with version 1.3, referred to page 22.
(https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_3/ug1414-vitis-ai.pdf)

ayeffkay, 2021-01-07:
> Thanks.
> What's your Vitis version? I tried with version 1.3, referred to [page 22](https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_3/ug1414-vitis-ai.pdf) for the explanation.

I tried Vitis-AI 1.3 from docker container `xilinx/vitis-ai:latest`. From the new version of docs it follows that your arch is supported by `vai_c_xir`, but I think the update was not completed by developers. And that's why this works in such a way :( For me it's a problem too, because our team has been trying to deploy model on Alveo U250. 
In any case, in the old version, only one DPU supported xir-based flow, and unfortunately, now models are compiled for the same DPU.

Hikaru-Furuta, 2021-01-07:
@ayeffkay
Which model are you trying to handle?
I was able to run several classification models succesfully.
If you wanna try these, I can provide you with the imformation about the procedure.

ayeffkay, 2021-01-09:
@Hikaru-Furuta Yes, it would be great, because I have no idea, what to do. We are going to quantize BERT, but first we decided to quantize toy regression model (from my issue [here](https://github.com/Xilinx/Vitis-AI/issues/239)). We adapted code from `resnet18_quant.py`, but after that the model was not compiled on any DPU (except `DPUCAHX8H`).

Hikaru-Furuta, 2021-01-11:
@ayeffkey
Sorry for the late reply
I just folloewed this page.
https://github.com/Xilinx/Vitis-AI/tree/master/examples/DPUCADX8G/tensorflow
hoping it's gonna help you

closed 2022-03-03 by qianglin-xlnx