Xilinx/Vitis-AI #238: Using quantized checkpoints after pytorch model calibration

ayeffkay, 2020-12-24:
I don't find in docs or somewhere else, is it possible to use checkpoints from calibrated model?
That is:

1. Run _torch_quantizer(quant_mode='calib')_
2. Get new _model.py_ and _quant_info.json_ (with int weights)
3. Run program next time and load _quant_info.json_ as model state dict (for not to quantize and calibrate again). I tried _torch_quantizer(module=pretrained_model, quant_mode='test', state_dict=quant_info.json),_ but it doesn't work.

As I see from examples, calibrated quantized model is not taken into account in the test mode, the test mode generates a new quantized model for deploy. What then is the point of model calibration if we do not use it after calibration? Any help would be appreciated :)

niuxjxlnx, 2020-12-24:
quant_info.json is not int weights, but tensors quantization step. 
model.py is a intermediate file which including network information but can not be used as input model file of vai_q_pytorch.

ayeffkay, 2020-12-24:
> quant_info.json is not int weights, but tensors quantization step.
> model.py is a intermediate file which including network information but can not be used as input model file of vai_q_pytorch.

But if I'll load model from model.py-it's just model architecture, without pretrained weights? And when I try to create instance of model class from _model.py_ and evaluate it, I get:
`AttributeError: 'NoneType' object has no attribute 'in_quant_part'`

niuxjxlnx, 2020-12-24:
model.py can not be loaded by torch_quantizer, which is a intermediate file which including network information.

ayeffkay, 2020-12-24:
> model.py can not be loaded by torch_quantizer, which is a intermediate file which including network information.

I don't load model.py with torch_quantizer: 
```
from quantize_result.FFN import FFN
model = FFN()
# data loader initialization here 
loss_fn = torch.nn.MSELoss().to(device)
loss = evaluate(model, loader, loss_fn)
```

And this produce
`AttributeError: 'NoneType' object has no attribute 'in_quant_part'`

niuxjxlnx, 2020-12-24:
model.py is NOT standard model file and can not be evaluted.

ayeffkay, 2020-12-24:
> model.py is NOT standard model file and can not be evaluted.
Ok. But is it possible to get checkpoints from calibrated quantized model? For example, in torch it's possible to save state dict or full checkpoint after quantization.

niuxjxlnx, 2020-12-24:
vai_q_pytorch has no interface function for this.

ayeffkay, 2020-12-24:
> vai_q_pytorch has no interface function for this.

Thanks a lot for your responses!

closed 2020-12-24 by ayeffkay