Xilinx/Vitis-AI #239: fast_finetune doesn't work on simple pytorch model

ayeffkay, 2020-12-25:
Hello! I'm trying **torch_quantizer.fast_finetune** for simple model. But I got the following error:

> Traceback (most recent call last):
  File "test.py", line 105, in <module>
    quantizer.fast_finetune(eval_loss, (quant_model, valid_loader, criterion))
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 76, in fast_finetune
    self.processor.finetune(run_fn, run_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 147, in finetune
    self.adaquant.finetune(run_fn, run_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/adaquant.py", line 445, in finetune
    net_loss = self.optimize_layer(node, float_layer, layer_inputs, layer_act_group, net_inputs, net_loss, last_quant_mods, device)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/adaquant.py", line 229, in optimize_layer
    _ = f_model(*new_input_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "quantize_result/NN.py", line 20, in forward
    self.output_module_4 = self.module_4(self.output_module_3)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/nn/modules/module.py", line 532, in __call__
    result = self.forward(*input, **kwargs)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/nn/modules/tanh.py", line 50, in forward
    fragpos = self.quantizer.get_bnfp(input_name, False)[1]
AttributeError: 'NoneType' object has no attribute 'get_bnfp'

Code to reproduce error:
```python
import numpy as np
from sklearn.datasets import make_regression
from tqdm import tqdm

import torch
from torch.utils.data import Dataset, DataLoader, SubsetRandomSampler
import torch.nn as nn
import torch.nn.functional as F

from catalyst.runners.supervised import SupervisedRunner
from pytorch_nndct.apis import torch_quantizer

class TorchDataset(Dataset):
    def __init__(self, X, y):
        super().__init__()
        self.X = X
        self.y = y
        
    def __len__(self):
        return len(self.y)
        
    def __getitem__(self, i):
        x = torch.from_numpy(self.X[i]).to(torch.float)
        y = torch.tensor([self.y[i]]).to(torch.float)
        return {'features': x, 'targets': y}

class NN(nn.Module):
    def __init__(self, input_size):
        super().__init__()
        self.l1 = nn.Linear(input_size, input_size * 3)
        self.l2 = nn.Linear(input_size * 3, input_size * 2)
        self.l3 = nn.Linear(input_size * 2, 1)
        
    def forward(self, *args):
        o1 = F.relu(self.l1(args[0]))
        o2 = torch.tanh(self.l2(o1))
        o3 = self.l3(o2)
        return o3
        
# DATA GENERATION
X, y = make_regression(n_samples=1000, n_features=3, n_targets=1)
data = TorchDataset(X, y)

idxs = np.random.choice(range(2), size=len(y), replace=True, p=[0.3, 0.7])
train_idx = np.where(idxs == 1)[0]
valid_idx = np.where(idxs == 0)[0]

train_sampler = SubsetRandomSampler(train_idx)
valid_sampler = SubsetRandomSampler(valid_idx)

batch_size = 64
train_loader = DataLoader(data, batch_size=batch_size, sampler=train_sampler)
valid_loader = DataLoader(data, batch_size=batch_size, sampler=valid_sampler)

# MODEL INITIALIZATION AND TRAINING
model = NN(X.shape[1])
optimizer = torch.optim.Adam(model.parameters(), lr=1e-1)
criterion = nn.MSELoss()

num_epochs = 32
loaders = {'train': train_loader, 'valid': valid_loader}
runner = SupervisedRunner()
runner.train(model=model, 
             optimizer=optimizer, 
             criterion=criterion, 
             loaders=loaders, 
             logdir='./logs', 
             num_epochs=num_epochs,
             verbose=True, 
             load_best_on_end=True)

targets = torch.cat([b['targets'] for b in valid_loader])
outputs = torch.cat([o['logits'] for o in runner.predict_loader(loader=loaders['valid'])])
loss = F.mse_loss(outputs, targets)
print('Validation loss', loss)

# QUANTIZATION
def eval_loss(model, loader, loss_fn):
    model.eval()
    epoch_loss = 0
    
    for i, batch in tqdm(enumerate(loader), total=len(loader)):
        x = batch['features']
        y = batch['targets']
        
        o = model(x)
        loss = loss_fn(o, y)

        epoch_loss += loss.item()

    return epoch_loss / len(loader)

quant_mode='calib'
rand_input = torch.randn(batch_size, X.shape[1])
quantizer = torch_quantizer(quant_mode, 
                            module=model, 
                            input_args=rand_input,
                            bitwidth=8, 
                            mix_bit=False)
quant_model = quantizer.quant_model
quantizer.fast_finetune(eval_loss, (quant_model, valid_loader, criterion))
loss = eval_loss(quant_model, valid_loader, criterion)
print('Validation loss after quantization', loss)
quantizer.export_quant_config()
```

qianglin-xlnx, 2021-01-05:
Hi @ayeffkay 
Thank you for your feedback. It's a bug. We will fix it.

qianglin-xlnx, 2021-03-17:
Hi @ayeffkay 
The bug has been fixed in VAI1.3.1.

closed 2021-03-17 by qianglin-xlnx