Xilinx/Vitis-AI #241: Vitis-AI 1.3 Caffe Compilation

jlamperez, 2020-12-27:
Hi,

I am trying to compile Vitis-AI 1.3 models using caffe conda environment inside the docker like this:

```
(vitis-ai-caffe) Vitis-AI /workspace/models > vai_c_caffe -p downloads/raw/cf_hourglass-pe_mpii_256_256_10.2G_1.3/quantized/deploy.prototxt \
> -c downloads/raw/cf_hourglass-pe_mpii_256_256_10.2G_1.3/quantized/deploy.caffemodel \
> -a arch.json \
> -o compile/hourglass-pe_mpii \
> -n hourglass-pe_mpii
```

And the output of this is:

```
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=None, layout='NCHW', model_files=['downloads/raw/cf_hourglass-pe_mpii_256_256_10.2G_1.3/quantized/deploy.caffemodel'], model_type='caffe', out_filename='compile/hourglass-pe_mpii/hourglass-pe_mpii_org.xmodel', proto='downloads/raw/cf_hourglass-pe_mpii_256_256_10.2G_1.3/quantized/deploy.prototxt')
[INFO] caffe model: downloads/raw/cf_hourglass-pe_mpii_256_256_10.2G_1.3/quantized/deploy.caffemodel
[INFO] caffe model: downloads/raw/cf_hourglass-pe_mpii_256_256_10.2G_1.3/quantized/deploy.prototxt
[INFO] parse raw model     :100%|█████████████████████| 157/157 [00:00<00:00, 351.77it/s]               
[INFO] infer shape (NCHW)  :100%|█████████████████████| 191/191 [00:00<00:00, 14927.74it/s]             
[INFO] infer shape (NHWC)  :100%|█████████████████████| 191/191 [00:00<00:00, 9337.62it/s]              
[INFO] generate xmodel     :100%|█████████████████████| 191/191 [00:00<00:00, 8465.64it/s]              
[INFO] generate xmodel: /workspace/models/compile/hourglass-pe_mpii/hourglass-pe_mpii_org.xmodel
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20201227-185027-1422"
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B2304_MAX_BG2
[UNILOG][INFO] Graph name: deploy, with op num: 457
[UNILOG][INFO] Begin to compile...
[UNILOG][INFO] Total device subgraph number 2, DPU subgraph number 0
[UNILOG][INFO] Compile done.
[UNILOG][INFO] The meta json is saved to "/workspace/models/compile/hourglass-pe_mpii/meta.json"
[UNILOG][INFO] The compiled xmodel is saved to "/workspace/models/compile/hourglass-pe_mpii/hourglass-pe_mpii.xmodel"
[UNILOG][INFO] The compiled xmodel's md5sum is 43a8a7a7b0bb258a6fa4a1e148adf969, and been saved to "/workspace/models/compile/hourglass-pe_mpii/md5sum.txt"
```

From this log I see that the DPU subgraph number is 0 what looks suspicious for me.

I have created a svg image from the xmodel using:

```
xir svg hourglass-pe_mpii.xmodel hourglass-pe_mpii.svg
```

And I see that the graph is different regarding to the xmodel downloaded for zcu102/zcu104 boards.

Someone knows if I am missing something in the `vai_c_caffe` command?

qianglin-xlnx, 2020-12-28:
Hi @jlamperez 
Please use the deploy.prototxt in the attachment and test it again. Thank you very much.

jlamperez, 2020-12-28:
Hi @qianglin-xlnx ,

Thank you very much, the prototxt that you have attached me for hourglass network solves the problem.

Now I would like to compile also the rest of caffe examples , because I am seeing that they are suffering the same problem, can you send me the modified prototxt (openpose, sp_net, ....)?

qianglin-xlnx, 2020-12-28:
Hi @jlamperez 
Thank you for your feedback.
We will fix this issue. Before we fix it, you can use the updated prototxt in the attachments. Also, you can quantize the model by yourself to get the correct prototxt. file. 
[openpose_deploy.prototxt.tar.gz](https://github.com/Xilinx/Vitis-AI/files/5747319/openpose_deploy.prototxt.tar.gz)
[sp_net_deploy.prototxt.tar.gz](https://github.com/Xilinx/Vitis-AI/files/5747320/sp_net_deploy.prototxt.tar.gz)

jlamperez, 2020-12-28:
Thank you very much @qianglin-xlnx for your support.

Best regards!

closed 2020-12-28 by jlamperez

AfifaIshtiaq, 2022-04-21:
> Hi,
> 
> I am trying to compile Vitis-AI 1.3 models using caffe conda environment inside the docker like this:
> 
> ```
> (vitis-ai-caffe) Vitis-AI /workspace/models > vai_c_caffe -p downloads/raw/cf_hourglass-pe_mpii_256_256_10.2G_1.3/quantized/deploy.prototxt \
> > -c downloads/raw/cf_hourglass-pe_mpii_256_256_10.2G_1.3/quantized/deploy.caffemodel \
> > -a arch.json \
> > -o compile/hourglass-pe_mpii \
> > -n hourglass-pe_mpii
> ```
> 
> And the output of this is:
> 
> ```
> **************************************************
> * VITIS_AI Compilation - Xilinx Inc.
> **************************************************
> [INFO] Namespace(inputs_shape=None, layout='NCHW', model_files=['downloads/raw/cf_hourglass-pe_mpii_256_256_10.2G_1.3/quantized/deploy.caffemodel'], model_type='caffe', out_filename='compile/hourglass-pe_mpii/hourglass-pe_mpii_org.xmodel', proto='downloads/raw/cf_hourglass-pe_mpii_256_256_10.2G_1.3/quantized/deploy.prototxt')
> [INFO] caffe model: downloads/raw/cf_hourglass-pe_mpii_256_256_10.2G_1.3/quantized/deploy.caffemodel
> [INFO] caffe model: downloads/raw/cf_hourglass-pe_mpii_256_256_10.2G_1.3/quantized/deploy.prototxt
> [INFO] parse raw model     :100%|█████████████████████| 157/157 [00:00<00:00, 351.77it/s]               
> [INFO] infer shape (NCHW)  :100%|█████████████████████| 191/191 [00:00<00:00, 14927.74it/s]             
> [INFO] infer shape (NHWC)  :100%|█████████████████████| 191/191 [00:00<00:00, 9337.62it/s]              
> [INFO] generate xmodel     :100%|█████████████████████| 191/191 [00:00<00:00, 8465.64it/s]              
> [INFO] generate xmodel: /workspace/models/compile/hourglass-pe_mpii/hourglass-pe_mpii_org.xmodel
> [UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20201227-185027-1422"
> [UNILOG][INFO] Compile mode: dpu
> [UNILOG][INFO] Debug mode: function
> [UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B2304_MAX_BG2
> [UNILOG][INFO] Graph name: deploy, with op num: 457
> [UNILOG][INFO] Begin to compile...
> [UNILOG][INFO] Total device subgraph number 2, DPU subgraph number 0
> [UNILOG][INFO] Compile done.
> [UNILOG][INFO] The meta json is saved to "/workspace/models/compile/hourglass-pe_mpii/meta.json"
> [UNILOG][INFO] The compiled xmodel is saved to "/workspace/models/compile/hourglass-pe_mpii/hourglass-pe_mpii.xmodel"
> [UNILOG][INFO] The compiled xmodel's md5sum is 43a8a7a7b0bb258a6fa4a1e148adf969, and been saved to "/workspace/models/compile/hourglass-pe_mpii/md5sum.txt"
> ```
> 
> From this log I see that the DPU subgraph number is 0 what looks suspicious to me.
> 
> I have created a svg image from the xmodel using:
> 
> ```
> xir svg hourglass-pe_mpii.xmodel hourglass-pe_mpii.svg
> ```
> 
> And I see that the graph is different regarding to the xmodel downloaded for zcu102/zcu104 boards.
> 
> Someone knows if I am missing something in the `vai_c_caffe` command?

Can you tell me why I face error no file found when conda tries to access model placed in my PC downloads

farah-rostom98, 2023-07-16:
hi i need to compile the resnet50 model and i get the same error using vitis AI 1.3 can you please send the deploy.prototxt file