Xilinx/Vitis-AI #244: "exec: \"/etc/login.sh\": permission denied": unknown

York1996OutLook, 2021-01-03:
(base) sm@dl4028:/etc$ sudo docker start f03db547386b
Error response from daemon: OCI runtime create failed: container_linux.go:349: starting container process caused "exec: \"/etc/login.sh\": permission denied": unknown
Error: failed to start containers: f03db547386b

hanxue, 2021-01-04:
Root is not required to run Docker containers - just make sure your user account (e.g. `sm`) is added to the `docker` group on the host. 

Could you try without `sudo`?

qianglin-xlnx, 2022-02-20:
Hi @York1996OutLook 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-20 by qianglin-xlnx