Xilinx/Vitis-AI #246: Pytorch quantization darknet model

david132048, 2021-01-06:
I'm trying to quantize darknet model. However, I got an error like below.

  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 88, in export_quant_config
    self.processor.export_quant_config()
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 157, in export_quant_config
    self.quantizer.export_quant_config()
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/quantization/torchquantizer.py", line 339, in export_quant_config
    self.organize_quant_pos()
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/quantization/torchquantizer.py", line 289, in organize_quant_pos
    shift_cut = fix_pos_w[-1] + fix_pos_i[-1] - fix_pos_o[-1]
TypeError: unsupported operand type(s) for +: 'NoneType' and 'NoneType'

I don't understand why it is happened. Anyone give me some solution?

niuxjxlnx, 2021-01-11:
Can you paste quant_results/quant_info.json here?

Tripton, 2021-02-03:
Hello,
I had exactly the same problem.
In my case it was because I forgot to enter data into the quantized model.
But this is necessary because, as far as I know, the quantized model must be calibrated first.
If this is not done, the model cannot be exported.
After using my test data for calibration, saving as XModel was possible.

niuxjxlnx, 2021-02-04:
Yes, a common reason to cause this error is calibration is not done before exporting quantization result.

vasdvczxc, 2021-02-22:
@Tripton 
Thank you for your help. 

However, I don't know how to calibration using test data. 

Can you share your tip?

Thank you.

dingguodong-826, 2021-03-04:
> 我正在尝试量化暗网模型。但是，出现如下错误。
> 
> 文件“ /opt/vitis_​​ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py”，第88行，位于export_quant_config
> self.processor.export_quant_config（）
> 文件中/vitis_​​ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py“，第157行，位于export_quant_config
> self.quantizer.export_quant_config（）
> 文件” / opt / vitis_​​ai /conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/quantization/torchquantizer.py“，行339，位于export_quant_config
> self.organize_quant_pos（）
> 文件” / opt / vitis_​​ai / conda / envs /vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/quantization/torchquantizer.py”，第289行，在organize_quant_pos
> shift_cut = fix_pos_w [-1] + fix_pos_i [-1]-fix_pos_o [-1]
> TypeError：+不支持的操作数类型：“ NoneType”和“ NoneType”
> 
> 在这里，fix_pos_w，fix_pos_i和fix_pos_o均为[8，None]。但是，当我使用resnet18模型运行时，它们的int值不是None。
> 
> 我不明白为什么会这样。有人给我一些解决方案吗？

I also encountered the same mistake as you when quantifying. Have you solved it?I had a problem executing this step
[image]

niuxjxlnx, 2021-03-05:
Please confirm calibration is done  before exporting quantization result. Calibration is done among model evaluation iteration.

niuxjxlnx, 2022-02-21:
Suppose the error is due to wrong model quantization script writing and clear error message will be reported in Vitis-AI 2.0. Will close this issue if no more response.

closed 2022-02-22 by niuxjxlnx